M5ez/M5ez #5: Display JPG Image

cslehel, 2018-09-04:
Hello, 

i have a minor problem and i hope you can guide me. 
I made a basic menu with 2-3 items. If i select one of them the m5stack will download an image to the SD card. 
I would like to show that image until i press a button (or a combination of 2 buttons). 
At this moment i am using the following command: 
M5.lcd.drawJpgFile(SD, filename); 
When the image stops loading the menu will show.

Thank you for considering my request.

ropg, 2018-09-05:
Yup, that should work. I haven't worked with `HTTClient.h` nor done anything with SD cards, so I can't judge if the rest would work, but wherever you place `ez.waitForButtons("OK");` it will draw a button and wait for it to be pressed.

closed 2018-09-05 by ropg

cslehel, 2018-09-05:
Thank you very much, i trust you but i cant try it right now :)