M5ez/M5ez #6: TaskScheduler

cslehel, 2018-09-06:
Hello, 

i ran into another problem. I would like to run a function every 10 seconds. 
I used before this library https://github.com/arkhipenko/TaskScheduler but i cant get to work correctly with M5ez. 
Without a menu but the M5ez library loaded the scheduler is working fine.
With myMenu.run(); in the loop the scheduler stops after the first call.
With myMenu.runOnce(); the scheduler works only when i select a menu item.

I would like to call a function every 10-15 seconds and update the menu items if it is possible.

I really appreciate any help you can provide.

ropg, 2018-09-06:
Why use a scheduler on top of a real multitasking OS like the FreeRTOS that is underneath on the ESP32? I really recommend creating true FreeRTOS tasks.

ropg, 2018-09-06:
That said: I am working on a way to include your own function into the loops of M5ez. And then in there you could see if 10 seconds have passed and do something. That is probably a few weeks out though.

Hmmm, maybe I could even do `ez.at(moment, yourFunction)` and `ez.every(milliseconds, yourFunction)` and keep a list of registered functions and when to call them. That would be spiffy and not too hard...

cslehel, 2018-09-06:
You are right, i got it working now. 
Thank you for your help.

closed 2018-09-06 by cslehel

ropg, 2018-09-08:
Good point. I will make this change affect the menu immediately. Cannot think of a quick workaround right now.

Should be back making some changes to M5ez sometime soon after I have put the finishing touches on [ezTime](https://github.com/ropg/ezTime).