M5ez/M5ez #12: ezTime.h missing?

cancodr, 2018-10-11:
I just upgraded to v2.0.0.  I had the previous version demo compiling and working ok.  I can't seem to get the current version compiled as the error comes back that "ezTime.h" is missing.

cancodr, 2018-10-11:
Sorry, I may be missing something really basic.  I see the M5ez.h file, but I don't see ezTime.h (which M5ez.h tries to include).

ropg, 2018-10-11:
> Sorry, I may be missing something really basic. I see the M5ez.h file, but I don't see ezTime.h (which M5ez.h tries to include).

Ah, my bad for missing that. You need to also install the ezTime library using the Library Manager, see the README, where it says "Repeat this process for the ezTime library". Alternatively, if you don't need the clock display in the header, you can also comment out the `#define M5EZ_CLOCK` line in M5ez.h.