M5ez/M5ez #14: M5stack fire: right LED is blue during demo

jeanvdr, 2018-10-12:
Don't know why, but the right LED strip on M5Stack fire is ON and blue with the m5ez-demo program.

jeanvdr, 2018-10-12:
we can add 
digitalWrite(15, LOW);
in the setup to correct this

ropg, 2018-10-13:
I don't have a fire yet, did all dev on M5ez on a grey M5Stack core. But I happened to order one yesterday, should be coming today. I'll test, put in the digitalWrite and release 2.0.1 sometime soon after.

ropg, 2018-10-13:
Just got my fire. M5ez works fine, but somehow the LEDs stay off with the demo program... Different versions of the hardware?

jeanvdr, 2018-10-14:
I don't use WPS so I didn't change anything on my M5stack library.

ropg, 2018-10-14:
It's not about WPS, it's to stop the ghost keypresses that (at least) some grey M5Stacks exhibit when using WiFi. But OK, good to know your M5Stack library is unmodified. Still waiting for at least one other person to have this problem before putting in a fix, and I'd like to understand what my code does to make it happen, because I don't touch any pins.

ropg, 2018-10-16:
Jean, can you try the OTA function in the demo app to get the OTA binary and then use it to OTA back to the demo app to see if the problem persists?

jeanvdr, 2018-10-16:
I will try that this evening

jeanvdr, 2018-10-19:
So, after trying several times, it seems the LED issue depends on the previous sketch loaded !!! and not on the m5ez demo itself, like a pin initialisation problem maybe due to the m5.begin procedure ?
For me, we can close this issue