M5ez/M5ez #15: Do you have any Arduino examples of using the clock and wifi in the headers

meteosat007, 2018-10-17:
Hi,

Really like the code and looks relatively simple yet I'm struggling to work out how to have a basic sketch that puts the clock, wifi and a title in the header. Have tried many variants of the commands but always get this error. Have managed several other sketches using M5Stack with wifi access, MQTT and various displays but would like to use your code as it looks cleaner. All help welcomed.

/var/folders/83/0n15kbkj0x90tdvm2gm489mc0000gn/T/arduino_modified_sketch_484742/z-sysInfo.ino: In function 'void ezTimeLog()':
z-sysInfo:39:28: error: expected primary-expression before numeric constant
   ez.header.insert(uint8_t 0, String "clock", uint16_t 40;
                            ^
z-sysInfo:39:38: error: expected primary-expression before string constant
   ez.header.insert(uint8_t 0, String "clock", uint16_t 40;
                                      ^
z-sysInfo:39:56: error: expected primary-expression before numeric constant
   ez.header.insert(uint8_t 0, String "clock", uint16_t 40;
                                                        ^
Multiple libraries were found for "SD.h"

My Sketch looks like this.

#ifndef MAIN_DECLARED

#include <M5Stack.h>
#include <M5ez.h>
#include <ezTime.h>

void setup() {
  ez.begin();
  ezTimeLog();

  while(true) {
    ez.yield();
  }
}

void loop() {

}

#endif  // #ifndef MAIN_DECLARED

void ezTimeLog() {

  ez.header.show("Sullivan Control");  
  ez.header.insert(uint8_t 0, String "clock", uint16_t 40;
  ezt::setDebug(INFO, ez.canvas);
  ezt::setInterval(20);

  #ifdef MAIN_DECLARED
    ez.buttons.wait("Abort");
    ezt::setDebug(NONE, Serial);
    ezt::setInterval(600);
  #endif
  
}

Thanks Alan

ropg, 2018-10-17:
Hi Alan,

First of all: You shouldn't put the variable types in there. They are there in thedefinition of the functions to show what the function expects, but in the actual call to the function you leave them out.

Furthermore the clock is already built into M5ez, so you cannot create a header widget called "clock" (or "title" or "wifi" for that matter). To see the clock, simply get online with your app: the clock should on by default.

closed 2018-10-17 by ropg

meteosat007, 2018-10-17:
Hi,

Ok yes can see that now.

I have the latest libraries in Arduino and have setup a very basic sketch with a header with my text in it and that loads into the unit and I see the dark blue header with the text I supplied but no clock or Wifi which is up and connected. Please what am I doing wrong here?

#include <M5Stack.h>
#include <M5ez.h>
#include <ezTime.h>
#include <WiFi.h>

const char* ssid = "Y";
const char* password = "***********";

const char* ntpServer = "de.pool.ntp.org";
const long  gmtOffset_sec = 3600;
const int   daylightOffset_sec = 0;

WiFiClient wifiClient;

void wifiConnect() {
    
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
   // Serial.print("*");
  }

  Serial.print("WiFi connected: ");
  Serial.println(WiFi.localIP());

}
void setup() {
  Serial.begin(115200);
  
  //M5.begin();
  ez.begin();
  ez.header.show("Sullivan Control");

  wifiConnect();

 //init and get the time
  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
  delay(1500);
}

void loop() {
  delay(1000);
}

Thanks Alan

meteosat007, 2018-10-17:
Ok so I have wifi bars in header, but still no clock. Do I need to begin the eztime or something else as the code below does not show a clock in header. :-(

  ez.header.show();
  ez.header.draw("clock");
  ez.header.draw("wifi");

meteosat007, 2018-10-17:
All sorted as it need a Canvas statement to make the clock appear...

Now to press on with the content.

Keep up the good work as it does make life easier now that I have the basics. 
Maybe a small example in the examples section that shows the clock and wifi in the header?

Cheers Alan

ropg, 2018-10-17:
Hi Alan,

Take a look at the M5ez demo program. You'll see it does absolutely nothing special to get clock and wifi in the header: they just work. So no need to provide ntp server, the SSID/password or really anything at all. Note that the `delay` function doesn't do enough: If you did it the way you do, you would want to run `ez.yield()`. But try this sketch instead:

```
#include <M5Stack.h>
#include <M5ez.h>

void setup() {
  ez.begin();
  ezMenu sc("Sullivan Control");
  sc.addItem("Settings", ez.settings.menu);
  sc.addItem("Something else", something_else);
  sc.run();
}

void loop() {
  // only here to keep arduino compiler happy:
  // program never leaves setup() because sc.run() never exits
}

void something_else() {
  ez.msgBox("Something else", "Indeed", "OK");
}
```

You will mostly want some menu item or something that points to `ez.settings.menu`. That way you have access to the settings to get online or change the clock settings or whatever.

ropg, 2018-10-17:
(If you surround your sketch or example code with three backticks ( \`\`\` ) on a line by themselves it will look nicer like like in the posts above...) Let me know if you get stuck and I'll try to help.