M5ez/M5ez #18: Timezone lookup for: GeoIP ... ERROR: Timeout

shellus, 2018-11-24:
Is there a way to specify a time zone in my source code?

ropg, 2018-11-24:
Whoops, had a server problem Should be solved now. Will make things more robust soon...

closed 2018-11-24 by ropg

shellus, 2018-11-25:
Is there a way to set the time zone in m5ez, for example to set the time offset directly.
@ropg

ropg, 2018-11-25:
> Is there a way to set the time zone in m5ez, for example to set the time offset directly.

Presently you can set timezone names (like "Asia/Beijing"), for which it then does a lookup. It's in the clock settings.

Rop

shellus, 2018-11-26:
I have set it to "Asia/Chongqing", but the log still shows `Timezone lookup for: Asia/Chongqing ... ERROR: Timeout`

ropg, 2018-11-26:
You can test my server by running (on a Mac or unix machine) `nc -u timezoned.rop.nl 2342`. Then just enter a query and end with Ctrl-D.

shellus, 2018-11-27:
```bash
root@pi:~# echo "Asia/Macau" |  nc -u timezoned.rop.nl 2342
ERROR Timezone Not Found
^C
root@pi:~#
root@pi:~# nc -u timezoned.rop.nl 2342
Asia/Macau
ERROR Timezone Not Found
ERROR Timezone Not Found^C
root@pi:~#

```

ropg, 2018-11-27:
It doesn't work with echo because it puts a carriage return after it:

```
$ nc -u timezoned.rop.nl 2342
Asia/MacauOK Asia/Macau CST-8
```

(Just type the "Asia/Macau" bit and then hit Ctrl-D.)

shellus, 2018-11-30:
是的，我证实了这一点
```
Hwn I try that, i get ERROR Timezone Not Found. Which makes sense when I look at the database. Below are the possible cities for Asia, Chongqing is not on the list.

Strangely I don't even see Beijing on there, but this is the database all unix machines I have access to use, so it has to make most people happy. Not sure what's up, but Macau seems to use CST (+8) too.

If you get timeouts, something else is up, not sure what. My server works...
```