M5ez/M5ez #19: Implementing OTA problem

skysingh, 2018-11-27:
I have been trying to use your tutorial on implementing OTA. So far - I created a respository on Github, published it and created a /compiled_binaries folder. I uploaded my compiled M5stack binary into this folder. So far so good. I got and ran the get_cert script on a linux machine. It generated the certificate but here was one difference from your tutorial. The resulting file was github_com.h instead of raw_githubusercontent_com.h. Anyway I continued by copying the github_com.h file into the sketch directory and modified the sketch to include the wifiupdate code as per your tutorial. Compiled and uploaded to M5stack. All good.
BUT when I run it finds the file but wifiUpdateError() gives Content-type must be "application/octet-stream" ,got text/html;charset=utf-8;

Any ideas what I did wrong ?

skysingh, 2018-11-27:
Update - I managed to get it work. OTA is nice in M5ez !!!
The problem is that I did not get the raw file url. Which is why I did not get raw_gitusercontent_com.h and so on. The trick was to get the raw URL. The View raw method did not work but this works ..
Click the file name in a GitHub repo.
Click Raw to display the file contents.
Copy the URL in your browser.
In the linux command line, run:
wget --no-check-certificate --content-disposition https://URL-from-step3/

.. it will display the raw url and you can use this with get_cert and the rest works as you documented.

ropg, 2018-11-27:
Glad you got it to work. What works for me is to use "Copy link location" on the "View Raw" link in github and give that to `get_cert`, I didn't need `wget`. `get_cert` uses curl to get the definitive URL.

closed 2018-11-27 by ropg