M5ez/M5ez #20: EZTime TZ a pain

skysingh, 2018-12-15:
I like EZTime but the time zone entry by city/country is a pain. Can you have an option to just input the timezone +/- hrs GMT ? Thanks. Also any hope for more documentation on how to use the widgets ?

ropg, 2018-12-15:
The first is more an ezTime than a M5ez issue, but alright...

The problem with just setting a UTC offset is that then you do not get transitions to and from Daylight Savings Time. The city format is used to ask the timezoned.rop.nl server (using `yourTZ.setLocation` for the POSIX format string for that timezone, which can contain information about DST change dates. But it doesn't have to, and you can also (using `yourTZ.setPosix`) set the POSIX information directly. For example: if you do:

```
Timezone India;
India.setPosix("IST-5:30");
Serial.println(India.dateTime());
```

You get a timezone that is at UTC +5:30 (the POSIX format uses the opposite sign from the usual format everyone else uses. (Yes, annoying). So doing `yourTZ.setPosix("XYZ+q") makes the timezone abbreveviation "XYZ" and fixes it at UTC-q hours. As explained [here](https://github.com/ropg/ezTime/#setposix) in the ezTime manual.  

As for widgets: this [thread](http://forum.m5stack.com/topic/262/m5ez-a-complete-interface-builder-system-for-the-m5stack-as-an-arduino-library-extremely-easy-to-use/76) on the M5Stack forum will probably help you.

closed 2018-12-15 by ropg

ropg, 2018-12-16:
I guess I didn't count on people wanting to do this manually when I documented. But M5ez does not use the default timzone, it uses `ez.clock.tz`, so you could do (I think) `ez.clock.tz.setPosix("XYZ-q")` to set XYZ timezone name at UTC+q hrs. Please test and let me know and I'll add it to the manual.