M5ez/M5ez #21: Numeric entries in textInput ?

skysingh, 2018-12-16:
Thanks for all your help - I am almost down to my last issue with M5ez in my application. I need to restrict data entry into the textInput box to only numbers. How can I do this ?

ropg, 2018-12-16:
Hmmm, I was thinking of building that and never got to it. The problem I had was that once you start going down that path, some people also want a comma, or a period, or a minus, or a colon and before you know it you have user defined keyboards. Which I could have done but would have taken a day or two to get right. And then I feared it would be one of those features that takes days to make and document and then nobody ever uses.

I guess the simple version of it would be to create a third argument to textInput that locks it to a given keyboard. Would everything from the NUM keyboard work for you? Either way, I'll leave this open and put it on the wishlist.

Orthodudz, 2019-07-23:
Hi @ropg , my added image for image menu won't show up. Is there any others things to do before the image file on SPIFFS to show?

percramer, 2020-04-27:
Somekind of numeric input would be very handy though.

vkichline, 2020-08-16:
Hello skysingh, percramer. Thanks for your input! We're preparing M5ez for a minor update which does not include this feature request, but I'm moving this to a product backlog that we will review before each release plan. This is a reasonable request I'm interested in as well, so we may very well include it soon.

closed 2020-08-16 by vkichline