M5ez/M5ez #24: Crashes with canvas.print..

skysingh, 2019-01-11:
I think there is some memory overflow problem with canvas.print. I had my program that uses M5ez crash after a period of time. Then reboot and repeat. I removed all the canvas.print statements and replaced them with M5.Lcd.print and did not have these odd crashes any more. Maybe you should take a look. Thanks.

ropg, 2019-01-11:
Whoops, yes. My bad. Now that I re-read that code: `canvas.print` creates a buffer in memory that stores every print until it has scrolled off the screen, that is the only way to do scrolling since the screen reading functions built into the display do not seem to work on the M5Stack. I should document that, and make it not store anything if scrolling is turned off, so people can make that tradeoff. keeping this issue open for now, will have new version out soon.

chb60, 2019-04-20:
Hello, this point is solved ? Because I have a worry with ez.canvas.print, every time i want to display something, the free memory head decrease, with other fonctionnality (canvas.pos, canvas.font) work well.
Only print, prinf lost memory at each call even canvas.scroll(false) is called
    ez.canvas.font(&FreeSans9pt7b);
    ez.canvas.scroll(false);
    ez.canvas.pos(10,ez.canvas.bottom()-50);
    ez.canvas.print("Temp *C "); -> this line lost memory every second in my loop
Thanks for your help

chb60, 2019-04-20:
I have try to to replace ez.canvas.print with M5.Lcd.print and no bytes lost, ESP.getFreeHeap() stay constant

ropg, 2019-04-20:
In order to facilitate scrolling, canvas.print stores everything that is printed until you clear the screen or until it scrolls off. I plan to disable scrolling by default and make more clear the limitations of printing if people need scrolling. Problem is that the display's native functions to support scrolling do not seem to work, so keeping in memory what is there is the only way.

chb60, 2019-04-20:
Hi Rob, no problem, I understand the worry, I can use M5 print in the canvas, that work fine too for my application
Thanks a lot and congrats for your M5ez

ropg, 2019-06-07:
In 2.0.1 scrolling is default off, and the possibility of crashes in some scenarios when you turn it on is documented in the readme.

closed 2019-06-07 by ropg