M5ez/M5ez #25: Unable to disable FACES without changing library

Instagraeme, 2019-01-15:
Would it be possible in future versions to define if FACES is enabled in the application code rather than in the library? As it stands at the moment the Wire library is always loaded and configures the I2C port. This means you cannot use that port for, for instance, a serial connection.

Otherwise the library is working really well for me. Thank you for all your work on it.

ropg, 2019-01-16:
What's wrong with commenting out the `#define M5EZ_FACES` from `M5ez.h`?

But I now see that I indeed do `Wire.begin()` even when FACES is turned off in the M5ez settings menu. I'm traveling and cannot test things easily right now. Could you edit `src/M5ez.cpp` take the `Wire.begin()` away at line 1003 and place it on a new line after `if (_on) {` (current line 1008) ?  Please tell me here if that does the trick, I will make sure that change is included in the next version.

What are you making, just out of interest?

Instagraeme, 2019-01-16:
The problem with changing M5ez.h is my memory! If I compile the code on a different machine than I normally use I need to remember to disable the M5EZ_FACES otherwise I cannot use the I2C port to read serial data. With PlatformIO this is trivial since the modified library can be kept with the application code. However, with the Arduino IDE it's more problematic. I can't check changing the position of Wire.begin() at the moment however looking at the code it would appear that it would fix the issue.

Projects I've written using your library so far:
1) Menu Interface to Home Assistant for turning on and off lights etc. 'Screensaves' after 30 seconds of inactivity to show electricity usage in the house. Menu shows the current status of each light.
2) Displaying the output from a set of scales to ensure a reading has been correctly recorded.
3) Diagnostic tool for decoding and displaying RS232/RS422 data read from various sensors. Also allows simulating sensors.
4) Update software stored on the SD Card to a device using the LAN module.

One thing I would like to do, which I haven't worked out yet, is how an addevent function running the background can invalidate a menu so that the menu is redrawn. For example so the addevent function can add or remove an item from the menu. Any advice?

ropg, 2019-06-11:
I moved the `Wire.begin()` to only be executed when FACES is on, and I created an `ez.redraw()` function to make sure a menu is redrawn if an event has changed the screen, so I am closing this issue. Let me know if there are additional issues.

closed 2019-06-11 by ropg