M5ez/M5ez #27: Is there a possibility of responding to smoothing fonts?

komikoni, 2019-01-17:
I like beautiful fonts.
Smoothing font is best for 320 x 240 screen.
The M5Stack library incorporates TFT_eSPI, but smoothing fonts can not be used at this time.
However, if you apply patches, smoothing fonts are available.

https://github.com/m5stack/M5Stack/issues/112

However, in M5ez which is essential to M5Stack, the display will collapse.

(Maybe M5Stack may correspond to smoothing fonts.) Is there any possibility that M5ez will correspond to smoothing fonts?

ropg, 2019-01-17:
Hey... Looks interesting. I should play with that at some point. First time I hear of these "smooth fonts", haven't seen anyone use them. I am struggling understand what you mean when you write "However, in M5ez which is essential to M5Stack, the display will collapse."

M5ez just uses TFT_eSPI through the M5Stack library, it doesn't do anything special. But it also doesn't load fonts from SD or SPIFFS, it expects them to be in memory. I could, I guess, make it so that when you specify a string instead of an address it will load a file by that name, but it's a big change because it's a lot of locations. I would need to be more convinced that lots of users want to use smooth fonts. 

How does one create these smooth fonts?

Related: did you check out https://rop.nl/truetype2gfx ?

vkichline, 2020-08-16:
Thank you for your suggestion! We are preparing a minor update release which does not include this enhancement, but it is strongly under consideration for a future minor or major release. I am closing the issue and moving it to the product backlog. We'll review this in future release plans.

closed 2020-08-16 by vkichline