M5ez/M5ez #29: WiFi: connecting without M5ez menu (ez.wifi.menu)

MacLe7400, 2019-02-13:
Hello,

I'd like to connect my m5stack to my wifi with out M5ez menu system.

I'm trying to use following sequence:

```
void setup() {
  ez.begin();
  [...]
  if (ez.wifi.indexForSSID("your-ssid") == -1) {
    ez.wifi.add("your-ssid", "your-key");
    ez.wifi.writeFlash();
  }
  [...]
}
```

But.. without results. There are no errors but there is no connection.

Full (example) code:

```
#include <M5Stack.h>
#include <M5ez.h>

const char* wifi_ssid = "myssid";
const char* wifi_password = "mypass";

void setup() {
  
  ez.begin();
  ez.header.show("WiFi test");

  ez.canvas.println("----> WiFi LIST");
  for (int8_t n = 0; n < ez.wifi.networks.size(); n++) {
    ez.canvas.print(ez.wifi.networks[n].SSID);
    ez.canvas.print(": ");
    ez.canvas.println(ez.wifi.networks[n].key);
  }
  ez.canvas.println("----> END");
  ez.canvas.println("");

  if (ez.wifi.indexForSSID(wifi_ssid) == -1) {
    Serial.println("WIFI: not found. Adding.");
    ez.wifi.add(wifi_ssid, wifi_password);
    ez.wifi.writeFlash();
  }

  ez.wifi.begin();

  ez.canvas.print("SSID:"); ez.canvas.println(WiFi.SSID());
  ez.canvas.print("Key:"); ez.canvas.println(WiFi.psk());
  ez.canvas.print("My IP:"); ez.canvas.println(WiFi.localIP().toString());

}

void loop() {
  // loop
}
```

First loop results correct ssid, password (my network).. however.. nothing happens.
  ez.canvas.print("My IP:"); ez.canvas.println(WiFi.localIP().toString());

results in My IP: 0.0.0.0
M5ez version: 2.0.1

Any clues?

Regards, M

ropg, 2019-02-14:
On first glance, I agree that that should work. Does the network you added show up in the menu if you afterwards put in the M5ez demo app?

Also, you can try compiling with `#define M5EZ_WIFI_DEBUG` and see what the serial output says, that should report all wifi-relevant things that are happening and might teach us something...

MacLe7400, 2019-02-14:
Unfortunately setting `#define M5EZ_WIFI_DEBUG` doesn't bring anything to `Serial` (simultaneously other `Serial.print `in code works perfectly).

However I managed how to use this conde to connect to WiFi simpy adding   `ez.yield()` to main `loop()` part and.. wait patiently (at least few seconds to connect - much longer that regular wifi library use).

So.. woking code should have at least something like:

```
void loop() {
  ez.yield();
}
```

However... I would really prefer to wait for the successful connection on ‘setup’ stage… something like:
`while ( ! WiFi.isConnected())  delay(500);`

But when I add such a loop to the code code just stops on this "while loop".

Consider full code (which stops on `while loop`):

```
#include <M5Stack.h>
#include <M5ez.h>

#define M5EZ_WIFI_DEBUG

const char* wifi_ssid = "my_ssid";
const char* wifi_password = "my_pass";

void setup() {

  ez.begin();
  ez.header.show("WiFi test");
  Serial.println("WiFi test");

  ez.canvas.println("----> WiFi LIST");
  for (int8_t n = 0; n < ez.wifi.networks.size(); n++) {
    ez.canvas.print(ez.wifi.networks[n].SSID);
    ez.canvas.print(": ");
    ez.canvas.println(ez.wifi.networks[n].key);
  }
  ez.canvas.println("----> END");
  ez.canvas.println("");

  if (ez.wifi.indexForSSID(wifi_ssid) == -1) {
    Serial.println("WIFI: not found. Adding.");
    ez.wifi.add(wifi_ssid, wifi_password);
    ez.wifi.writeFlash();
  }

  ez.wifi.begin();

  while ( ! WiFi.isConnected()) delay(500);

  ez.canvas.println("CONNECTED");
  ez.canvas.print("SSID:"); ez.canvas.println(WiFi.SSID());
  ez.canvas.print("Key:"); ez.canvas.println(WiFi.psk());
  ez.canvas.print("My IP:"); ez.canvas.println(WiFi.localIP().toString());

}

void loop() {
  ez.yield();
}

```

How can I check if wifi is successfully connected and THEN initiate main loop?

MacLe7400, 2019-02-21:
Hello, Any clues how can I resolve this issue?

LordInchcape, 2019-04-22:
There is some Black Magic going on with M5Stack and WIFI connection. In Arduino on ESP32 the call to WiFi.begin()  will always work. But the same code on M5Stack usually does not work. As the original append says when you use the M5Ez menu system WIFI connection works, but when you try manually directly from your sketch it appears not to work.
I have seen several references to M5Stack and WIFI connection problems but no clarity on a solution.
Did you get a resolution to this MacLe7400?

deival1980, 2019-05-30:
You can try that:
`....
ez.wifi.begin();

  while ( ! WiFi.isConnected()) 
{
delay(500);
ez.yield();
}

  ez.canvas.println("CONNECTED");
.....`

ropg, 2019-06-11:
I cannot reproduce this issue, but It looks from the comments like this is fixed, or at least a workaround is known. So I am closing it. Let me know if anyone thinks a change to M5ez is nevertheless called for.

closed 2019-06-11 by ropg