M5ez/M5ez #35: Repeatedly restarts when updating with OTA.

HigashidaTomoyuki, 2019-05-20:
If you update with `OTA_https` of `examples`, reboot will be repeated. I do not know the solution. Also, please let me know if there is a way to do OTA using Bluetooth.

closed 2019-06-25 by HigashidaTomoyuki

laplanteliberee, 2020-04-04:
Same issue here?
Any solution? how to stop it or reset it? Cannot connect with arduino IDE

ropg, 2020-04-04:
Strange. I'll get my M5Stack out and experiment at some point, but this used to work for me. Does the serial output show anything? It should show the unit booting at least, and then possibly a stack trace...

laplanteliberee, 2020-04-04:
It constantly reboot, cannot access it anymore... not enough time to go to configure wifi
Serial monitor failed to connect to /dev/TTYUSB0

laplanteliberee, 2020-04-04:
I managed to load another program (and stop it rebooting)
simply holding down button A and reset button until Arduino ide attempted to connect
then releasing it and allowing it to upload without a chance for the old code to run