M5ez/M5ez #37: Printing degree symbol?

andyrbarrow, 2019-06-05:
This may not specifically be for M5ez, but I'm using the library right now to create a position display for my boat. I need to be able to print a degree symbol (raised "o" character) as part of a display of latitude and longitude. The only font I can find that includes it is TomThumb, which is too small for what I need.

Anyone have any ideas?

ropg, 2019-06-05:
> This may not specifically be for M5ez, but I'm using the library right now to create a position display for my boat. I need to be able to print a degree symbol (raised "o" character) as part of a [...]

You could leave the space blank and draw or print separately (kinda ugly hack, but might work), or try to convert a font using my truetype2gfx tool at https://rop.nl/truetype2gfx/ .

andyrbarrow, 2019-06-05:
Oohhh, convert a font! I'd forgotten about doing that.

I'll give it a shot. Thanks a million for a great library.

ropg, 2019-06-05:
You could leave the space blank and draw or print separately (kinda ugly hack, but might work), or try to convert a font using my [truetype2gfx tool](https://rop.nl/truetype2gfx). The TomThumb works (I found a TrueType font file, but it doesn't seem the optimal font for information presentation.

[image]

closed 2019-06-05 by ropg