M5ez/M5ez #41: Unclear how addEvent works, only called once

HighTML, 2019-06-10:
I want to use M5ez (with disabled EZ Wifi) in conjunction with EspMQTTClient.
Standalone EspMQTTClient example connects to Wifi and to MQTT Broker. In M5ez application I need to call espMQTTClient.loop();    I think this should be done with events.. however, it looks like the event wrapper is called only once:

```
EspMQTTClient client;

void setup() {
#include <themes/default.h>
  ezt::setDebug(INFO);

  ez.addEvent(clientLoopWrapper);
  ez.begin();

  Serial.println("SETUP complete");
}

uint16_t clientLoopWrapper() {
  Serial.println("\nInside clientLoopWrapper");
  client.loop();
}

void loop() {
  ezMenu mainmenu("Welcome");
  mainmenu.txtSmall();
  mainmenu.addItem("Send", send_msg);
  mainmenu.addItem("Built-in settings", ez.settings.menu);
  mainmenu.upOnFirst("last|up");
  mainmenu.downOnLast("first|down");
  mainmenu.run();

}

```

In the serial output there is only 1 appearance of "Inside clientLoopWrapper". So the loop() of EspMQTTClient is called just once...

andyrbarrow, 2019-06-10:
I am also using MQTT. I found that you need to create a loop in each menu instance which you intend to have MQTT polling in. In my case, I have a single button on the screen that takes the user back to the main screen.

In the main loop I have a menu entry for Wind:

`void loop() {
  ezMenu mainmenu("HeyYA Info System");
  mainmenu.txtBig();
  mainmenu.addItem("Location", location_display);
  mainmenu.addItem("Wind", wind_display);
  ... whatever else ...`

...and the first part of the menu instance looks like this:
`void wind_display(){
  ezMenu windDisplay;
  ez.header.show("Wind");
  ez.buttons.show("Main Menu");
... some code to prep the screen, like another setup()  ... (in my case, drawing a compass rose)

  String btnpressed = ez.buttons.poll();
... and this is like another loop() ...
  while (btnpressed != "Main Menu"){
    btnpressed = ez.buttons.poll();
      client.loop(); // this is the call to MQTT
      String btnpressed = ez.buttons.poll(); // this exits the loop and returns to the main menu
      .... whatever the rest of your looped code is ....
}
`

HighTML, 2019-06-10:
I always need a working MQTT connection for incoming messages. This means I have to manually call ```client.loop()```  in all the places where I wait for user input. And, I cannot use the blocking M5EZ calls.

The addEvent(...) was created for this purpose, to be able to call your own code while in the EZ refresh loop. But, I can't get it to work..  Any suggestions @ropg ?

ropg, 2019-06-10:
Whoops, I forgot to document that returning 0 from the event function will delete the event (fixed). So just returning 1 should do the trick. Internally the event handler subtracts one from the number of milliseconds, so returning 1 still means 'as often as possible'. If you notice performance issues, you may want to find the optimal number of milliseconds between each execution of a given event function.

closed 2019-06-10 by ropg