M5ez/M5ez #44: #include "images.h"

twegithub, 2019-08-30:
I am new to all this and barely know what I am doing.  Your work with M5ez is amazing and incredibly useful.  My IDE compilation hangs up due to the lack of library on the following line:
#include "images.h"

I cannot find where to add this library.  

Can you please help?

ropg, 2019-08-31:
The only place I can find where any of my code includes an `images.h` is in the m5ez-demo code. That comes in its own directory, which includes that file. 

The easiest way to try the demo app is to select it from the File / Examples / M5ez menu in the Arduino GUI. Otherwise please tell me exactly what you are doing to get this error...

twegithub, 2019-08-31:
Rop,
That worked!  I reloaded the M5ez -demo code and it picked up the images.h
library.  All good.

You have written a valuable menu system for the M5STACK.  It is very useful
and I thank you so much for helping me.  Great contribution!

Best wishes.
Tom

*Tom Evernham  W9GBA*
*Spruce Creek Fly-In, Port Orange, FL*
  (summer - Sylvan Heights Mountain Community, Brevard, NC)
VoIP: 386.304.8233    Cell:  386.679.9702
tomevernham@gmail.com
[image: Picture]

twegithub, 2019-08-31:
Rop,
That worked!  I reloaded the M5ez -demo code and it picked up the images.h library.  All good.

You have written a valuable menu system for the M5STACK.  It is very useful and I thank you so much for helping me.  Great contribution!

Best wishes.
Tom

ropg, 2019-09-01:
I'm glad you enjoy using M5ez...

Have fun,

Rop

closed 2019-09-01 by ropg