M5ez/M5ez #45: Different color map on newer M5stack

skysingh, 2019-09-09:
I have been using M5ez on M5stack. Works great. But I just received a new batch of M5stack and the colors are mapped differently. For example magenta is now green, blue is yellow. What gives here and how to fix? I need the program to maintain compatibilty with the older M5's.

ropg, 2019-09-09:
Oh wow. I think that would a) really suck and b) break a lot of other things. 

Can you tell me as precisely as you can in what context this is seen. And do you have any clue how many units this affects?

Rop   ( - sent from phone - )

skysingh, 2019-09-09:
Rop:

 

Test program is attached. Also screenshot of old and new M5stack. Very worrisome. I have sent the same information to the M5stack people.

 

Regards,

 

-          Vijay Singh

ropg, 2019-09-09:
I’d be inclined to give this a little time and see what they did and what they’re going to do about it. Sound like a screwup to me. How is everything else on these M5’s? Also in wrong colors?

Rop   ( - sent from phone - )

skysingh, 2019-09-10:
Hello Rop:

 

This is what I got from M5stack but it does not work. Will keep asking them what is going on.

 

-          Vijay Singh

 

From: teresa@m5stack.com [mailto:teresa@m5stack.com] 
Sent: Monday, September 09, 2019 10:38 PM
To: 'Vijay'
Subject: RE: You sent a payment

 

Hi Vijay,

 

1.    Sorry to make you confused. It caused by our new IPS screen. The newly-produced M5Core replaces the screen with better display performance and higher viewing angle, so it has some compatibility problems with the old Arduino library. When using the old library for screen driving, it will produce reverse color display. You can open the Arduino. The library management option will upgrade your M5Stack library to the latest version (after 0.2.8) to solve this problem. Attached photo for your reference.

 

2.    About the ‘cannot program via USB port’ problem, how many devices has this problem ? Did they work normally when you receive them ? We may need more detail info to confirm the problem clearly. Thanks!

 

 

------------------------------------

Best Regards,

 

Teresa Pan | Key Account Manager

 

M5Stack Technology Co., Ltd.

Add: RM106, Block A, MingLiang Technology Park, Zhuguangbei Road, Nanshan, Shenzhen, China.

Tel: +86 755 88600970

Wechat/Whatsapp: +86 13631292780

Email:  <mailto:teresa@M5Stack.com> teresa@M5Stack.com

Website:  <http://www.m5stack.com/> www.m5stack.com

 

From: Vijay <vsingh@art-engineers.com> 
Sent: Monday, September 9, 2019 9:36 PM
To: teresa@m5stack.com
Subject: RE: You sent a payment
Importance: High

 

Hello Teresa:

 

 

We have a serious problem with the lot of 100 M5stack you sent recently

 

1.	The color map seems to be altered. Magenta now shows as green. Blue is yellow and so on. Can you have somebody in tech support address this problem. It is serious because we cannot maintain program compatibility with previous M5stack. Maybe there is some way to revert the color map back to the old one.
2.	A few of the new M5stacks are refusing to connect over USB so we cannot program them. What is your policy on warranty replacement?

 

Thanks. We are very disturbed about the color problem because we use the M5stack in a OEM product and we expect to buy over 1000 next year.

 

Regards,

 

*	Vijay Singh
*	vsingh@art-engineers.com

ropg, 2019-09-10:
Ok, so that means it’s out of scope for M5ez, this is for the underlying library to deal with. Good luck.

closed 2019-09-11 by ropg