M5ez/M5ez #46: Scheduling tasks with longer periods

woyzeck, 2019-12-09:
Hi,

I'd like to schedule a rather costly background task for execution every three hours (the exact time doesn't matter - I just don't want to waste resources by setting off the function more frequently). As far as I understand {{void ez.addEvent(uint16_t (*function)(), uint32_t when = 1)}} takes a 16-bit uint as the argument for the period, which limits the schedule to at most 65535 milliseconds or just over a minute.
Any chances that this is changed to uint32 ?

Kind regards
Nikolaus

ropg, 2019-12-09:
I had planned to build something like alarms, given that it has a built-in clock. That way one could schedule something for days out, or every third Wednesday, you know the drill.

Then later that bit would optimally come with a UI component people can also set their own events,create PDA-like functionality, etc. 

It would then have one of the underlying M5ez events just briefly check every minute whether one of these events was due.

Just haven't gotten round to it.

If you case, just fire an M5ez event every minute and have it check whether 3 hours have past. That work?

woyzeck, 2019-12-09:
Hi Rop,

Thx for your superfast response. I was thinking of such mechanism, though it seems to be awkward. Anyway, it‘ll work.

Regards
Nikolaus

closed 2019-12-10 by ropg