M5ez/M5ez #49: Menus blocking

skysingh, 2020-01-13:
I have been incorporating blynk into my sketch that use M5ez. One issue I have is can a menu be nonblocking? I have the user in a menu and blynk times out because the menu blocks execution. I tried to use timer and also ez.yield() but no use.

So not really a M5ez issue but would appreciate a workaround.

ropg, 2020-01-13:
I'm not familiar with blynk. But I imagine that blynk can be happy as long as something is ran periodically. If so: using `ez.addEvent` you can set a function to run periodically and then use `ez.redraw()` at the end if that function might have changed the screen.

Just to give you an example:

```
uint16_t eventFunction() {
	// Insert here whatever blink needs to be ran periodically

	ez.redraw();	// Only if screen has changed
	return 100;	// To run every 100 milliseconds
}
```

And then before you call menus or other blocking things, you call

`ez.addEvent (eventFunction);`

And if you don't want to run the event anymore, simply call

`ez.removeEvent (eventFunction);`

(or have the event routine return 0).

Let me know if this works...

skysingh, 2020-01-14:
Thanks for the reply. Blynk is kinda magic so I cannot figure what exactly needs to run periodically. One work around would be have the menu timeout if there is no button pressed for a time. This would ensure that blynk cannot be stopped forever. Blynk will recover automatically once the program runs again. Do you think is possible to implement a menu timeout?

 

*	Vijay

ropg, 2020-01-14:
Read a tiny bit about Blynk. Have you tried putting `Blynk.run();` in the eventFunction above?

skysingh, 2020-01-16:
Thanks for your help. That idea worked. But I still have a problem that I cannot time a menu out. Even with runOnce(), it waits for the user to select something. I have everything running in the background using events but I can’t exit the menu without direct user interaction. Really need a menu timeout or a workaround.

 

Thanks,

 

*	Vijay

ropg, 2020-01-16:
First of all note that uou can clear the screen and display whatever you want from the event routine, and then return to the menu later, just call `ez.redraw()` before returning then.

That said: I can see there are scenarios where you want to exit the running menu.

A menu ran with runOnce returns the number of the option selected. I would prefer to have a way to make the menu exit with 0 if some command is called, probably from the event routine. That way we prevent flicker: the menu is not timed out all the time but only exits if we know for sure that something else needs to be displayed. And then your code can see that was an irregular exit by just looking at the value returned. Something like 'ez.menuExit()' sets a global flag which then makes any running menu exit immediately. (RunOnce() exits with value zero, run() just exits). If you really wanted it to just time out you could set an event for that and just call ez.menuExit from there. 

Would this work for you?

(The problem with regular timeouts is that you really want to put the cursor back where it was if you go back into the menu, otherwise the menu selection becomes jerky...)

ropg, 2020-01-16:
The question is also whether this also makes other things like msgBoxes and textInputs exit... I'd have to think about this a bit more.

ropg, 2020-01-16:
Is your code online somewhere for me to look at?

skysingh, 2020-01-16:
Rop:

 

It would work but ez.menuExit() does not exist(right?). I do not need to renter the menu where I left off. If the user wants the menu again it starts at the top just as in the first invocation.

 

*	Vijay

skysingh, 2020-03-30:
Hello Rop:

 

My new problem is quite serious and I do not know if you can be of help but it is worth a try. I have a fairly complex sketch built around your excellent M5ez. I added Blynk to it so get remote access and it worked fine. Now in testing we are find that if WiFi is lost for some reason Blynk refuses to reconnect automatically. By deleting code I found that if I have ez.begin() in the setup code it will NOT reconnect. However if this is missing or even if use M5.begin() it works perfectly and reconnects automatically.

 

Any possible clue on what might be the problem. I am using large program minimal SPIFF and OTA. I suspect that the Blynk app is storing some auth codes in flash and ez.begin() is clobbering it.

 

Months of work is about to go down the drain!!

 

Thanks,

 

*	Vijay Singh

ropg, 2020-03-30:
The code for `ez.begin()`:

```
void M5ez::begin() {
	m5.begin();
	ezTheme::begin();
	ez.screen.begin();
	ez.settings.begin();
}
```

If really just calling `ez.begin()` is enough to cause your problem, you could replace it by calling these three other things yourself and just leave a different one off each time to narrow it down. I have very little time right now: we're working on a COVID-related [project](https://covid-at-home.info).

skysingh, 2020-03-30:
Hello Again Rop:

 

This is where Blynk seems to be storing codes. May be it gives you a clue why M5ez is interfering.

 

*	Vijay

 

struct ConfigStore {

  uint32_t  magic;

  char      version[15];

  uint8_t   flagConfig:1;

  uint8_t   flagApFail:1;

  uint8_t   flagSelfTest:1;

 

  char      wifiSSID[34];

  char      wifiPass[64];

 

  char      cloudToken[34];

  char      cloudHost[34];

  uint16_t  cloudPort;

 

  uint16_t  checksum;

} __attribute__((packed));

 

ConfigStore configStore;

 

const ConfigStore configDefault = {

  0x626C6E6B,

  BOARD_FIRMWARE_VERSION,

  0, 0, 0,

  

  "",

  "",

  

  "invalid token",

  "blynk-cloud.com", 443,

  0

};

 

#include <Preferences.h>

Preferences preferences;

 

void config_load()

{

  memset(&configStore, 0, sizeof(configStore));

  preferences.getBytes("config", &configStore, sizeof(configStore));

  if (configStore.magic != configDefault.magic) {

    DEBUG_PRINT("Using default config.");

    configStore = configDefault;

    return;

  }

}

 

bool config_save()

{

  preferences.putBytes("config", &configStore, sizeof(configStore));

  DEBUG_PRINT("Configuration stored to flash");

  return true;

}

 

bool config_init()

{

  preferences.begin("blynk", false);

  config_load();

  return true;

}

 

void enterResetConfig()

{

  DEBUG_PRINT("Resetting configuration!");

  configStore = configDefault;

  config_save();

  BlynkState::set(MODE_WAIT_CONFIG);

}

 

template<typename T, int size>

void CopyString(const String& s, T(&arr)[size]) {

  s.toCharArray(arr, size);

}

ropg, 2020-03-30:
Hi there... From first glance they *seem* to be storing in their own space, which should not interfere: `preferences.begin("blynk", false);` 

I have no clue, really... And never played with Blynk... Sorry...

skysingh, 2020-03-31:
Thanks for your help. I have narrowed it down the M5ez event timers. If I delete any addEvents, and instead implement through SimpleTimer Blynk stays happy.

 

*	Vijay

bEse11, 2020-05-06:
Hi Rob,

Congrats for these libraries, both M5ez and ezTime are fantastic. Visually most impressive M5 interface I've seen so far.

I tried hard to make it work, but the blocking code makes me nuts. Specially when I try to keep run my loop() as fast as possible for sensor readings and external libraries. 1 millisecond event loops mostly not enough. I hope you consider refactoring the library to remove delays.
Until then, I use M5Stack TreeView - not nearly as fancy visually as your Interface, but it works without blocking.

Keep up the good work!
Cheers, Andras

ropg, 2020-05-06:
Glad you like it, and sorry it doesn't presently suit your needs.

So you just need to run an event more often than once per millisecond? What does your code need to do? Or do you need to somehow break out of the menu and make M5ez do different things based on events outside of M5ez? The latter is actually quite difficult conceptually, because of the way you program for M5ez, which is fairly linear. 

Thinking about faster events, how about a way to just have an event that fires whenever M5ez is done doing its thing, and then you just yield back to M5ez by returning and get control back as soon as M5ez is done. I'd have to look at the code again to see where to put that in.

bEse11, 2020-05-09:
Interacting with other machines/sensors can be time sensitive: high baud rate serial, SPI, or any two-way communication, when the other party waits for you to interact. It would be useful to see an ez.addEvent example in the demo sketch, maybe I'am just missing something.

M5ez is a library designed for interact with humans. Human-machine interaction can be as slow as 3-4 frame/sec ( 250ms), perfectly bearable from a microcontroller (except animations, 20 fps-50ms).
And there is only one peer (1 human).
If you dealing with machines, there can be multiple peers, with pretty tight timing requirements, so priority is on this side of the code.

Well, my code(s) could be implemented as a callback function, that returns to M5ez, and it may solve the timing issue. If this is a short fix, then it would be great to see it in the coming versions.
Library methods (for sensors, or just for the mentioned blynk library) tied to the loop(), still has to be called explicitly, or/and needs the mentioned libraries to be refactored to work with M5ez.

Personally, I favor the M5Stack guys' style: They left the loop() for the users, and just asked them to call M5.update() in each loop() . M5.update() handles the display, buttons, built in sensors, etc. (maybe wifi as well, but not sure). 
It's kind of a low-level MVC, separates the main program logic from the UI, and M5ez basicly a very well equipped, eye-catchy UI building tool. (In my eyes: M5ez 3-button keyboard won the 2019 Esp32 Oscar award in "Best keyboard solution" category... at least my vote is yours ;) ).

You set the bar high with M5ez, hope someone takes it as a good example and makes a more loop() friendly UI library.
I know I am a little late to this party, since the library really matured, with lots of functions implemented in this "linear" way. It is perfect to interact with it in human level speeds, and it's suitable for most of the makers.

P.S.: ezKeyboard would be fantastic as a separate library :)

vkichline, 2020-08-16:
skysingh, bEse11, we are preparing for a minor release and are closing issues and moving suggestions to the product backlog. The issues you mention here are of concern to a number of uses, so we'd like to consider them carefully and address them in a future major release. We will keep this issue on the backlog and review it carefully in our design for 3.0.

closed 2020-08-16 by vkichline

skysingh, 2020-12-07:
Hey Rop:

 

Hope all is well with you.

 

I have been shifting most of my new projects to M5Atom and M5stickCplus. So I am not using M5ez but there is one feature I really need – That is the cool OTA you developed for M5 stack. Do you think this feature could be adapted to the Atom?

 

Thanks,

 

*	Vijay Singh

ropg, 2020-12-08:
I would suggest you take a look at how it works and just copy the code you need. It's not all that complicated, and there's no reason it shouldn't work on the Atom. One of the tricky bits is including the right SSL certificate, but I think that's well-explained.

skysingh, 2021-03-26:
Hello Rop:

 

Hope all is well with you. Quick question – I am trying to use the faces keypad. I followed your instructions in m5ez and I can see the keypresses in the textInput box but for the life I do not know how to get the data in a String I can use. Any examples?

 

Thanks,

 

*	Vijay Singh