M5ez/M5ez #50: Slow connect to Wifi after boot

cristian-spiescu, 2020-02-03:
Hi,

This is my first issue to this repo. So I'd like to begin by congratulating @ropg for this immensely time saving lib! 

And back to my issue. I have a saved Wifi network + autoconnect ON. After booting, it takes more than 1 minute to connect to Wifi. I share below my findings. During my investigations I found 2 items: **A)** a strange behavior of `WiFi.begin()` and **B)** something that I consider a glitch in the code/state machine.

Here is a debug log:

Lib version 2.1.2. Working w/ platform.io.

```
EZWIFI: Starting scan
EZWIFI: Scan got 10 networks
EZWIFI: Match: wifi-home-dachau, connecting...
[1] EZWIFI: Connect timed out...
[2] EZWIFI: Scan got 10 networks
[3] EZWIFI: No (further) matches, waiting...
EZWIFI: Starting scan
EZWIFI: Scan got 12 networks
EZWIFI: Match: wifi-home-dachau, connecting...
EZWIFI: Connected, returning to IDLE state
```
## A) a strange behavior of `WiFi.begin()`

[1]: happens almost always. Even if I increase the timeout (e.g. to 100s). So from a reason that I don't understand this line systematically fails:

https://github.com/ropg/M5ez/blob/147cb67c1b48a6424401a420c52cebd7c76d2126/src/M5ez.cpp#L1507

The ESP seems to be in some kind of state that makes this always fail at the beginning (maybe because a scan is in progress?). Subsequent retries work w/o problems. I did some trial/error modifications, w/o success. A new, simple sketch executes `WiFi.begin()` pretty quickly.

## B)  something that I consider a glitch in the code/state machine

And speaking of subsequent retries. Immediately after [1], happens [2]. It finds networks again, but this time it doesn't connect. And it enters [3], i.e. waiting mode for 1 min.

The reason for this is line:
https://github.com/ropg/M5ez/blob/147cb67c1b48a6424401a420c52cebd7c76d2126/src/M5ez.cpp#L1528

Commenting this, leads to a quicker connect. Because the wait doesn't happen any more. However, I don't understand what was the reason of this line. It seems to have a purpose. So just commenting/eliminating it may cause issues in other scenarios.

Anyway, even w/ this fixed, **A)** still happens which is pretty strange.

Thanks a lot in advance.
Best regards,
Cristian

cristian-spiescu, 2020-02-03:
Update regarding issue **A)**. Weird discovery: changing the line:

https://github.com/ropg/M5ez/blob/147cb67c1b48a6424401a420c52cebd7c76d2126/src/M5ez.cpp#L1509

by modifying 7000 to 2000, i.e. a smaller timeout, makes the connection almost instantaneous. 

From my understanding of the code, this kind of modification should have had no impact on whatsoever.

vkichline, 2020-07-23:
Thanks for your research on this! Sorry it has laid idle for some time; I am investigating and will get back to you soon.

cristian-spiescu, 2020-07-23:
Thanks a lot!

vkichline, 2020-07-24:
The problem that I'm seeing is that the first time the ESP32 tries to connect to a stored SSID, it throws a Event: 5 - STA_DISCONNECTED, Reason: 203 - ASSOC_FAIL. This happened immediately, but our code still waited 7 seconds for a timeout, then tried other SSIDs. If I set the state machine back to scanning mode immediately when this happens, it reconnects almost instantly. This may be different than your problem, though.
Right now it's in a separate repo (https://github.com/vkichline/M5ez/tree/slow_connect) if you'd like to try it. Or you can wait until we complete the PR and code review (which would be safer) but it would be good to know if this addresses your issue.

vkichline, 2020-07-24:
Minutes after completing the PR, I found a case that makes this the wrong approach, but I have another idea. I will post another PR soon.

vkichline, 2020-07-25:
The current PR seems low risk to me. If its the same issue you're experiencing, it should be fixed.
The reason commenting out `_current_from_scan++` worked for you was that it caused a repeat try (after a seven second timeout) of the zone that experienced the spurious disconnect. However, if you had an incorrect password registered or an access point that's just not connecting at the moment, you'd fail to log in completely. This fix reties almost immediately on the disconnect w/ ASSOC_FAIL.

vkichline, 2020-08-12:
This has been fixed. Let us know if there are any further problems. The release has not been updated yet, so at this point you'll have to manually download the library to get the new change. New release coming soon.

closed 2020-08-12 by vkichline

cristian-spiescu, 2020-08-13:
Thanks a million!