M5ez/M5ez #51: 'class SDLib::SDClass' has no member named 'cardType'

sysdl132, 2020-02-06:
when I compile "m5ez-demo",I got a error messege like this:
```
'class SDLib::SDClass' has no member named 'cardType'
```

sysdl132, 2020-02-06:
first I compile a m5ez code with z-sysinfo.
```
exit status 1
'class SDLib::SDClass' has no member named 'cardType'
```
then I compile m5ez-demo(with no change)
still:
```
exit status 1
'class SDLib::SDClass' has no member named 'cardType'
```

sysdl132, 2020-02-06:
[image]

ropg, 2020-02-06:
Oh crap. That means they changed something about the SD card interface, and now the code doesn't work anymore. Sucks.. I will try to find time soon to fix it. (But it will be march, I'm afraid..)

(Anyone up for fixing this?)

sysdl132, 2020-02-07:
i only have a way to upload it.
disable the SD from the demo and remove SD library.

sysdl132, 2020-02-14:
And before compiling,I disabled a line in SD/src/utility/Sd2pinmap.h:(Line 524)
```
#error Architecture or board not supported.
```
That means maybe ESP32 board is not supported.

USING:Arduino SD Library
==========================
There is an another question:Can you support M5StickC or StickV?

sysdl132, 2020-02-14:
I also suggest you can import some labels.like:

[image]

sysdl132, 2020-04-03:
I update the SD library and the problem fixed.Thanks.

sysdl132, 2020-04-03:
===Do not write under this line===

closed 2020-04-03 by sysdl132