M5ez/M5ez #52: Program crashes when logging data

posaidi, 2020-03-09:
hello

at the moment im working on a project where i use the M5Stack to get data from a different board and save it in a cloud every few seconds

. but my program crashes everytime when i try to execute a function where i try to log the data on thingspeak.com while i'm in a menu or submenu.

the function im trying to execute:

//----------------------------------------------------------------------------------------------
int writeThingSpeak(){ //Log Data on ThingSpeak.com
//----------------------------------------------------------------------------------------------
ThingSpeak.setField(1,state);
ThingSpeak.setField(2,volt);
ThingSpeak.setField(3,uamp);
leistung = ((volt * uamp)/1000000);
ThingSpeak.setField(4,leistung);
ThingSpeak.setField(5,t_gas);
ThingSpeak.setField(6,t_box);
ThingSpeak.setField(7,vers_m);
ThingSpeak.setField(8,vers_s);
ThingSpeak.writeFields(myChannelNumber, myWriteAPIKey);
}

ropg, 2020-03-09:
No idea, but just some random pointers in hope that they'll help.

* You can create your own events to run periodically while the menu is blocking. In case ThingSpeak needs something to run periodically to keep from crashing you could run that from an event.

* There is a [plugin](https://github.com/me-no-dev/EspExceptionDecoder) for the Arduino IDE to analyse the ESP32 crash info to see what its call stack looked like. (If that's what you mean by "my program crashes")

Otherwise too much depends on what your code is doing and what ThingSpeak does... For further debugging.

posaidi, 2020-03-11:
problem solved. i added an own event wich runs periodically where i execute the function. thank you!

closed 2020-03-11 by posaidi