M5ez/M5ez #53: How to update data displayed on canvas?

posaidi, 2020-03-11:
Hello Rop

I'm trying to display some values in a submenu-item. Now I'm updating those values every 6 seconds with an event. but even if I redraw the canvas at the end of the event-function, the values in my menu won't change. So i wanted to ask if there's another way to refresh the canvas so that it displays the new values?

ropg, 2020-03-11:
The menu will not change because it is running with the values that were given to it when it was started. The canvas refresh is thought more for when you did something special (display a dialog box, whatever) and then return from your event to the regular programming. Events should have a way to interrupt "regular programming" and restart the menu or something. But things get quite hairy quickly when multiple strands of programming  try to communicate through the same user interface. 

I'm presently working on an information site for Covid patients, which has to take priority, but will leave the issue open to remind me to give events and menus some more think when I get back here...

posaidi, 2020-03-23:
Solved the problem by adding another event, wich draws "textboxes" on the canvas every few milliseconds.

closed 2020-03-23 by posaidi