M5ez/M5ez #54: Buttons triggered automatically after connected to wifi?

sindney, 2020-04-02:
I uploaded the demo project to my m5stack, and i found the physical buttons will be randomlly triggered after connected to wifi ?? Is there anyone having the same issue here? Or it's just my hardware's issue. I'll write a button demo with wifi on to see if it's the hardware's fault later.

sindney, 2020-04-03:
I wrote a simple test, and it works fine, it seems it's not the hardware's issue?
```c
#include <M5Stack.h>
#include <WiFi.h>

const char* ssid     = "xxxx";
const char* password = "xxxx";

// The setup() function runs once each time the micro-controller starts
void setup() {
  // init lcd, serial, but don't init sd card
  M5.begin(true, false, true);
  M5.Power.begin();
  M5.Lcd.clear(BLACK);

  Serial.println();
    M5.Lcd.print("Connecting to ");
    M5.Lcd.print(ssid);
    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        M5.Lcd.print(".");
    }

    M5.Lcd.println("WiFi connected");
    M5.Lcd.println("IP address: ");
    M5.Lcd.println(WiFi.localIP());
}

// Add the main program code into the continuous loop() function
void loop() {
  // update button state
  M5.update();
 
  // if you want to use Releasefor("was released for"), use .wasReleasefor(int time) below
  if (M5.BtnA.wasReleased()) {
    M5.Lcd.print('A');
  } else if (M5.BtnB.wasReleased()) {
    M5.Lcd.print('B');
  } else if (M5.BtnC.wasReleased()) {
    M5.Lcd.print('C');
  }
}
```

sindney, 2020-04-03:
And i just found that if i build the program with mac (clang) there's no this "automatic button triggering" problem, it just happens when i built it on windows, i'll confirm this later, if it's the compiler's problem, i'll close this issue and keep using clang.

sysdl132, 2020-04-03:
But my device(on windows 10) is no problem?

sindney, 2020-04-03:
> 
> 
> But my device(on windows 10) is no problem?

I'll try again on my windows 10 after work.

sysdl132, 2020-04-03:
> > But my device(on windows 10) is no problem?
> 
> I'll try again on my windows 10 after work.

Oh,I'm using Arduino Pro on windows.It means the compiler is clang.

sindney, 2020-04-03:
> 
> 
> > > But my device(on windows 10) is no problem?
> > 
> > 
> > I'll try again on my windows 10 after work.
> 
> Oh,I'm using Arduino Pro on windows.It means the compiler is clang.

I'm using visual studio 2019 on windows and i reproduced this bug, guess it's related to compilers. I'll switch to clang for now.

closed 2020-04-03 by sindney

sysdl132, 2020-04-03:
(I'm using vs 2019 and visual micro,too)

sysdl132, 2020-04-03:
=====do not write below this line=====

sindney, 2020-04-03:
> 
> 
> (I'm using vs 2019 and visual micro,too)

额，我好像知道了，我下的esp32的hardware是https://gitee.com/bitplay/arduino-esp32这里的，我在另一台windows上也是编出来会自动按按钮，我试试clone github上的
直接用espressif那个下载太慢根本下不下来
我在公司的mac是espressif上面下的，烧的程序就没问题，估计是这个问题

sindney, 2020-04-03:
我最后手动在github上把所有最新的esp32 board相关文件手动下载摆放，打出来的包就没问题
和编译器无关，只是国内镜像版本问题估计

ropg, 2020-04-03:
I edited in the translations. Let's try to keep this is English, even if we have to use computer translations. I highly recommend [DeepL](https://www.deepl.com/translator). Fewer languages, but so much better than [Google Translate](https://translate.google.com/).

sindney, 2020-04-04:
> 
> 
> I edited in the translations. Let's try to keep this is English, even if we have to use computer translations. I highly recommend [DeepL](https://www.deepl.com/translator). Fewer languages, but so much better than [Google Translate](https://translate.google.com/).

got it, i'v translated my answers to english.