M5ez/M5ez #55: Image menus?

skysingh, 2020-04-03:
For the life of me I cannot figure out the ez images menus work. In the demo example there are a number of images in the single images.h file. How were these converted from jpeg and combined ?

ropg, 2020-04-03:
Ha... You found something that I did indeed not document. At the time I wrote M5ez, it was something a lot of people did and code circulated widely to turn BMPs and JPGs into header files. There was C code as described [here](http://forum.m5stack.com/topic/49/lesson-1-2-1-lcd-how-to-create-image-array) and there's [this](https://javl.github.io/image2cpp/) I just googled.

Please test this out, IIRC I just took my images from some other program that already had a neat image menu. I will leave this issue open as I need to add this to the documentation.

kornel-schrenk, 2020-05-02:
> Oh. I had figured out how to convert the jpeg to C array. The part I do not understand how you put several jpeg in one images.h file and how did you specify the names for each image (For example – wifi_jpg) so you can AddItem each one?

I hope this helps: https://schrenk.hu/2020/04/29/M5StackImageFormats/ . I also added a new PR to M5ez, which contains an example - called M5images, which demonstrates how to use various image formats in an image meu.

paultech, 2020-07-14:
Image loader support is being added currently with our PR being a useful reference.

closed 2020-07-14 by paultech