M5ez/M5ez #58: auto generate CSR certifite

sysdl132, 2020-04-09:
I must manually generate CSR every time.
Can you set it to auto generate a CSR?
```
(ez.wifi.update("web_addr", >CSR<, &progress_bar)     // auto generate CSR
```

ropg, 2020-04-09:
I don't understand... Where you say "CSR" is actually the spot to point to a root certificate, not a Certificate Signing Request. This is because the ESP32 is a little too lightweight to have a complete list of valid root certificates like browsers do, so you have to provide it.

The page [here](https://github.com/ropg/M5ez/tree/master/examples/OTA_https) is also pointed to from the user manual for `wifi.update` and it shows you how to use the [`get_cert` script](https://raw.githubusercontent.com/ropg/M5ez/master/tools/get_cert) that is in the the tools directory of the repo to get both the actual download URL and the root certificate for a website. Include the .h file that the script saves, include it in your project and use `root_cert` as the root certificate pointer.

When I say "actual URL" I mean that some URLs forward you to different URLs a few times before you actually download the file. The script will tell you what the effective URL is: use that URL.

You need to change your .h file if the SSL certificate used by the website you download from derives from a different root certificate. (Like when they change certificate provider, say)

sysdl132, 2020-04-09:
Thanks for your answer.My OTA problem now solved.

closed 2020-04-09 by sysdl132