M5ez/M5ez #61: Timezone could use suggestion

technerdchris, 2020-04-16:
Having to enter a posix timezone was a bit non-intuitive, perhaps a suggestion: Example: +4 or IST-5:30

Outstanding work, BTW. You made me look forward to menus. 1st time in 28 years.

mohir, 2020-04-23:
Hello,
I do not manage to get the timezone right :(
need IST (Israel, gmt +2),
entering IST does not work,
entering gmt+3 or just +3, shows etc/GMT+3 but the time is GMT-2
please advise.
Amazing work you've done! Thank you

ropg, 2020-04-23:
The plusses and minuses in the POSIX format are the other way from what you would expect. See the [manual](https://github.com/ropg/ezTime/#timezones-1) which says:

> Hours offset from UTC, meaning subtract one hour from this time to get to UTC. (Note offset is often written elsewhere the other way around (so +1 in this case), just to confuse things.) Could also specify minutes, like -05:30 for India.

As for what to enter for Israel: the code is Asia/Jerusalem, but for some reason my time server does not supply a POSIX code for that. Will look into this as soon as possible...