M5ez/M5ez #62: Text flickering

percramer, 2020-04-19:
Hi,

i am using 

`M5.Lcd.setTextColor(WHITE, BLACK);
ez.canvas.printf(" PM 1.0 : %2.2f \r\n", pm1);`

but my value keeps flickering. Like it is not using the background color?

Any idea?

ropg, 2020-04-19:
Internally, that does (among other things):
```
	m5.lcd.setTextColor(_color);
```
where `_color` is the current print color. The documentation for m5.lcd says "If backgroundcolor is not given, current background color is used". A certain amount of flicker is unavoidable if you print one value over another: it first has to clea the area before it can print. What you could do is store the value that is displayed in a variable and then only print it if it is different from that value (and then store that value in the variable). That way you at least kill all the flicker when the value is constant.

percramer, 2020-04-19:
I think I must change the title a bit since flickering isn't the right word. It doesn't seem to clear the background of the old text. So if I write my sensor value to the same spot (and the sensor value changes every 1 to 5 seconds) then after a while there is just a white spot left. Shouldn't the background pixels be filled with black pixels (if I pass a foreground and background color to SetTextColor)?

for now I have solved it by drawing a black rectangle first

` ez.canvas.print(" PM 1.0 : ");`
` M5.Lcd.fillRect(ez.canvas.x(), ez.canvas.y(), 320, 18, BLACK);`
 `ez.canvas.println(pm1,2);`

ropg, 2020-04-19:
Ah... That is a different and surprisingly complicated issue when you have proportional fonts with ligatures and potential overhang both ways. AdaFruit, the makes of the fontlibrary, have a [good text](https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts) on it. Quoting from that:

>One “gotcha” to be aware of with new fonts: there is no “background” color option…you can set this value but it will be ignored. This is on purpose and by design.
>
>The background color feature has typically been used with the “classic” font to overwrite old screen contents with new data. This only works because those characters are a uniform size; it’s not a sensible thing to do with proportionally-spaced fonts with characters of varying sizes (and which may overlap).
>
>To replace previously-drawn test when using a custom font, use the getTextBounds() function to determine the smallest rectangle encompassing a string, erase the area using fillRect(), then draw new text.

Long story short: drawing a box is the right approach, and if you want to get fancy there are functions to figure out exactly how big the box would need to be, but you are fine with a static location it seems...

closed 2020-04-19 by ropg