M5ez/M5ez #67: Can't call ez.msgBox from inside an event

vigonotion, 2020-05-23:
I'm trying to call a msgBox inside an event, but as soon as I do that, the screen flickers and the ESP restarts.

This is my code:

```c++
#include <M5ez.h>

void setup() {
	ez.begin();
}

void loop() {
  ez.addEvent(updateCube, 3000);
	ez.msgBox("M5ez minimal program", "Hello World !", "Settings");
	ez.settings.menu();
}

uint16_t updateCube() {

  ez.msgBox("cuti", "Congratulations! You solved the cube.");

  return 0;
}

```

This is my error:

```
M5Stack initializing...OK
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4008a57a  PS      : 0x00060833  A0      : 0x8008c40e  A1      : 0x3ffafe50  
A2      : 0x800d44e8  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060820  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffafe50  
A10     : 0x00000003  A11     : 0x00060823  A12     : 0x00060820  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x800d44e8  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x4008a57a:0x3ffafe50 0x4008c40b:0x3ffafe80 0x4008ca7f:0x3ffafea0 0x40085404:0x3ffafec0 0x40085439:0x3ffafee0 0x400858fd:0x3ffaff00 0x4000beaf:0x3ffaff20 0x40158b9f:0x3ffaff40 0x400d44e5:0x3ffaff60 0x400d46ab:0x3ffaff90 0x400d7371:0x3ffaffe0 0x400d7d66:0x3ffb01f0 0x400d14bd:0x3ffb0260 0x400d3184:0x3ffb02d0 0x400d31dc:0x3ffb02f0 0x400d7c6b:0x3ffb0460 0x400d7e91:0x3ffb0490 0x400d14bd:0x3ffb0500 0x400d3184:0x3ffb0570 0x400d31dc:0x3ffb0590 0x400d7c6b:0x3ffb0700 0x400d7e91:0x3ffb0730 0x400d14bd:0x3ffb07a0 0x400d3184:0x3ffb0810 0x400d31dc:0x3ffb0830 0x400d7c6b:0x3ffb09a0 0x400d7e91:0x3ffb09d0 0x400d14bd:0x3ffb0a40 0x400d3184:0x3ffb0ab0 0x400d31dc:0x3ffb0ad0 0x400d7c6b:0x3ffb0c40 0x400d7e91:0x3ffb0c70 0x400d14bd:0x3ffb0ce0 0x400d3184:0x3ffb0d50 0x400d31dc:0x3ffb0d70 0x400d7c6b:0x3ffb0ee0 0x400d7e91:0x3ffb0f10 0x400d14bd:0x3ffb0f80 0x400d3184:0x3ffb0ff0 0x400d31dc:0x3ffb1010 0x400d7c6b:0x3ffb1180 0x400d7e91:0x3ffb11b0 0x400d14bd:0x3ffb1220 0x400d3184:0x3ffb1290 0x400d31dc:0x3ffb12b0 0x400d7c6b:0x3ffb1420 0x400d7e91:0x3ffb1450 0x400d14bd:0x3ffb14c0 0x400d3184:0x3ffb1530 0x400d31dc:0x3ffb1550 0x400d7c6b:0x3ffb16c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
M5Stack initializing...OK
```

tommyteck, 2020-06-23:
I have the same issue it appears that the ez.yield retriggers the current task and creates the crash.

vkichline, 2020-08-13:
Hello vigonotion;
Sorry about the long delay with your issue; we have put together a team for handling these now.
I have submitted Pull Request #96 to address this issue. I think it's a solid and low-risk change, but if you would like to test the change and provide feedback before it's checked in, it's available from this fork: https://github.com/vkichline/M5ez/tree/bug_reentrant_event
I appreciate your providing a minimized repro case! It made the first steps of validation very easy.

vkichline, 2020-08-14:
We've checked in the change now. There has not been an official release, so at this point you will have to download the library manually to get the latest changes.

closed 2020-08-14 by vkichline