M5ez/M5ez #68: [Feature Request] Ordered menus

vkichline, 2020-05-26:
It would be nice to have optionally ordered menus. For example, inserting each file name from an SD without acquiring and sorting the entire list before building the menu would be 'ez'-er.
The simplest and least disruptive approach might be to add an optional function pointer to the ezMenu c'tor:
`ezMenu(String hdr="", int8_t(*orderingFunction)(String intro, String existing) = NULL)`
This would allow the user-defined function to deal with the `name` or the `caption`, alphabetically or numerically, as it wishes. I see that `leftOf` and `rightOf` are exposed, so this should be quite adequate.
Note that if the orderingFunction were replaceable:
`ezMenu.setOrderingFunction(int8_t(*orderingFunction)(String intro, String existing))`
features like reversing the menu order would be ez as well.
Alternately, making strategic functions virtual and data protected rather than private would allow specialization by inheritance, but it seems like both documentation and support could be significantly more complicated.

ropg, 2020-05-26:
Hmmmm... You sound like pull request instead of a feature request material. :)

(Otherwise it's gonna take some time...)

vkichline, 2020-05-26:
I'm happy to give it a shot. Just wanted to see how you felt about the approach first.
The callback signature will be `const char*` 's, not Strings.
Do you prefer code and docs in the same PR?

ropg, 2020-05-26:
Did I tell you M5ez was my first project in C++ ?

I prefer keeping things simple, so anything that makes doc or support difficult is best avoided. Keep in mind that most users are Arduino tinkerfolk that use it because they can get an interface really easily. These are (by and large) not expert C++ coders. 

Why const char* instead of Strings?

And maybe _only_ have `.setOrderingFunction` to keep the documentation looking simpler for people that don't need it?

vkichline, 2020-05-26:
I will provide `const char*` to the compare function both to clarify the intent (the user should not modify the values provided) as well as for safety (the user cannot modify the values without wrecking havoc, and an inexperienced user might consider it an opportunity to slip in a hack.)
I think you're right about leaving the c'tor alone. It would be quite consistent with the API to add the ordering function as one sets the buttons, font, etc. after creating the menu on the rare occasions that its even needed.

ropg, 2020-05-26:
Sounds good. Yes, code and docs in same PR please... :)

vkichline, 2020-05-27:
PR submitted (#70.) It was a pleasure working on the codebase. Clean and consistent.

closed 2020-05-27 by vkichline