M5ez/M5ez #69: Polling , reading buttons and timing out

mega-byte, 2020-05-26:
Hi Rop and first can I say what an awsome tool M5ez is. You've done a great job and made life for us mere users much easier. I don't know if my issue is a bug or my ignorance but here goes......

I’m in all sorts of trouble trying to set up a function with the following characteristics:
Display a screen with a line or two of instructional text ending with press OK to continue.
At the same time display a button saying “OK”.
When OK has been pressed, go on to the meat of the function which is to immediately record a reference pressure from a bme pressure sensor, then go into a loop which every  approximately 3 seconds:  reads the pressure from the sensor , calculates a height corresponding to the pressure difference between the current and reference pressures, and displays it by refreshing the M5 screen. The exit from the loop is when you press the Done button whose caption is of course displayed as soon as you go into the loop.

I’ve left my example code with the remnants of debugging instrumentation I’ve tried but either I’m using the code in a way that was not intended or there’s something odd about the way in which ez.buttons.wait() and ez.buttons.poll() work .

On a similar tack I would appreciate an elegant example of how to perform functionality in a loop until either a button is pressed “Done” or a timer expires. I’ve tried various approaches using the ez tool but I can’t claim any success.

void htMeas() {
  //ez.msgBox("Height Gauge","Press OK to zero the height gauge","",false);
  ez.canvas.clear();
  ez.canvas.font(&FreeSansBold9pt7b);
  ez.canvas.y(ez.canvas.top() + 100);
  ez.canvas.print(String("   Press OK to zero height gauge"));
  ez.buttons.wait("OK") ;
  //Serial.println("OK received to zero the ht gauge - now delay before continuing....");
  delay(500);   //tried various values here.....
  ez.buttons.show("Done"); 
  float refPressure = (bme.readPressure()/100);
  String butstr = "dummy initialisation string" ; 
  int htTmpStart = (millis()-100000) ;    // Set big to ensure we read immediately on first entering the do loop
  do {
    if (millis() > (htTmpStart + 3000)) {  
                                          //Don't try to read too frequently to allow the sensor to cope
      getHtNow(refPressure) ;
      htTmpStart = millis();
    }
    butstr = ez.buttons.poll();
    Serial.print("  Poll of buttons gives: ");
    Serial.println(butstr);
    //delay(100);   // tried various delay values here.... 
  } while (butstr != "Done"); 
}

void getHtNow(int refPressure){
  pressureHeight = int (((bme.readPressure()/100) - refPressure) * 27.3053) ;
  Serial.printf("\n\rHeight: %3.0f ft", pressureHeight);
  //delay(5000);     //  to see whats going on
  ez.canvas.clear();
  ez.canvas.font(&FreeSans18pt7b);
  ez.canvas.y(ez.canvas.top() + 75);
  ez.canvas.print(String("       Height  "+String(pressureHeight,0)+" ft"));
  ez.buttons.show("Done"); 
  }

 Can you please offer any guidance?

technerdchris, 2020-06-12:
@mega-byte I have similar problem trying to interact with this amazing body of code. IMHO, it is difficult to understand how to "do something" at the same time as the menu is present. Almost every GUI I use has a means of continuously showing status(es) and pressing buttons changes the "machine state" of my software. Not being able to see how to accomplish this has caused me to begin designing my own UI for the M5Stack, which is supremely disappointing as m5ez is so well done and my result will be lots of work, won't be flexible/reusable, and won't look as nice.

ropg, 2020-06-13:
With events you have a way to run things in the loop of M5ez. Can you be as specific as possible as to what you would want? How would you like it to work?

ropg, 2020-06-13:
Also please consider that for presenting changing information, maybe an M5ez menu isn't the best way. You could display a top bar and buttons and print something to the canvas yourself. That way you have more flexibility.

vkichline, 2020-08-16:
Hello, mega-byte. We are preparing for a minor release, so I am closing this issue but retaining it on our product backlog. We plan to address design issues like you mention here in future major releases, and your suggestions will be taken into account in our 3.0 design. Thanks for your input!

closed 2020-08-16 by vkichline