M5ez/M5ez #71: Suggestion: invert the usage of feature inclusion defines

vkichline, 2020-05-31:
Rop; I appreciate the use of defines to omit entire feature areas if desired, such as:
```
// (From M5ez.h)
#define M5EZ_CLOCK
#ifdef M5EZ_CLOCK
    ...
#endif
```
...etc. However, to utilize these, I must modify the library header file itself.
If instead you inverted the usage, like:
```
// (Proposed for M5ez.h)
// #define OMIT_M5EZ_CLOCK
#ifndef OMIT_M5EZ_CLOCK
    ...
#endif
```
...then I could include the desired define(s) in my code to cherry-pick features without modifying the library itself:
```
// (My code)
#define OMIT_M5EZ_CLOCK
#include <M5ez.h>
...
```
This would simplify taking library updates and sharing my code with other users. In addition, it would have no impact on existing code already using M5ez.

ropg, 2020-05-31:
Have you tested? I don't think that is true: the libraries are built separately, and defines you put in your code have no effect on that cycle of the build. I think there's a way with `#pragma gcc` directives, but haven't looked into it.

vkichline, 2020-05-31:
I have not in fact tried it yet. I will give it a shot and see if I can confirm that the generated code is omitted and not just access to it.

vkichline, 2020-05-31:
We all have a bad day some days. :-)
On the positive side, I can now prove that you are right.
I will close this.

closed 2020-05-31 by vkichline