M5ez/M5ez #72: PR submitted for sorted menus (#70.) I've been using this a wekk with no problems.

vkichline, 2020-06-05:
PR submitted (#70.) It was a pleasure working on the codebase. Clean and consistent.

_Originally posted by @vkichline in https://github.com/ropg/M5ez/issues/68#issuecomment-634396882_

paultech, 2020-07-15:
Think we can redo this PR now that we have cleaned up. I liked the simple implementation.

Should we offer some sane default sort functions?  ie: `M5Ez::sort_alphabetical_asc`  `M5Ez::sort_alphabetical_desc`

vkichline, 2020-07-15:
Sort functions are in the example program. Ascending/descending case sensitive/insensitive name/caption sorting.
However, I find that simply if ezMenu's 'private' section is changed to 'protected', I can implement sorting by subclassing. This keeps M5ez from bloating, and I may find other enhancements I want to add but aren't of universal interest.
Moving everything to 'protected' may be overkill, but keeping everything 'private' isn't helpful for the customer. I will experiment to see what the minimum change would be.

paultech, 2020-07-15:
We don't use polymorphism anywhere else in the project. I wouldn't expect users too for sorting. I think the previous implementation was the cleanest that matches our current API.

vkichline, 2020-07-15:
I agree, I like the look of `M5Ez::sort_alphabetical_asc`.
Having terrible computer problems, I'm switching to my laptop now, will add the change when I am functional again.

vkichline, 2020-07-16:
Resubmitted due to computer failure; #86.

vkichline, 2020-07-21:
PR checked in

closed 2020-07-21 by vkichline