M5ez/M5ez #73: Losing preference settings

vkichline, 2020-06-27:
Hello, Rop!
I've been enjoying using M5ez a lot, but I had found the preferences seemed to be unreliable, except for the WiFi networks, which were always rock solid. I believe I tracked this down today.
In M5ez.cpp, in ```ezWifi::writeFlash()```, line 1211 is: ``` prefs.clear() ```.
I suspect this is a holdover from early days where networks were the only persistent settings. Now, however, this removes my other preferences.
I think what's intended is to specifically erase the networks. I have substituted the following for the beginning of the method, and seems to have solved my preferences issues.

```
    void ezWifi::writeFlash() {
        Preferences prefs;
        String idx;
        uint8_t n = 1;
        prefs.begin("M5ez", false);
        prefs.remove("autoconnect_on");
        while (true) {
            idx = "SSID" + (String)n;
            if(!prefs.remove(idx.c_str())) break;
            idx = "key" + (String)n;
            prefs.remove(idx.c_str());
            n++;
        }
        ....
```

vkichline, 2020-07-10:
Thanks, Paul. I’d like to address this one first, because it affects most people and is simple.
I have a noob procedural question:
I forked M5ez a while back, made some changes, and merged them to master. I wish I had kept them in my own fork, but that boat has flown.
It don’t see any way to create another clean fork.
What’s the best way to submit just this delta, leaving the doc updates for second and sorting until later?

Assigned #73<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fropg%2FM5ez%2Fissues%2F73&data=02%7C01%7C%7Ca9eaaa452e714951fc5908d8251479c0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637300117151282698&sdata=iMC829hlvB7GVGyN%2F0VeTg5NpUcPNN6MnWzT4FYBbhg%3D&reserved=0> to @vkichline<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fvkichline&data=02%7C01%7C%7Ca9eaaa452e714951fc5908d8251479c0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637300117151292706&sdata=xBsz2wER%2Fj8r0u%2B9HsrEAd1SNt0idyLUqz1qzVAHpLU%3D&reserved=0>.

vkichline, 2020-07-12:
PR submitted, #80.

paultech, 2020-07-12:
Fixed by #80

closed 2020-07-12 by paultech