M5ez/M5ez #75: How do I add header to Images menus?

pipy71, 2020-07-02:
Hello, 
Sorry, I am a newbie to programming this and so please forgive me if this is not the correct place to ask this sort of question.

How do I add the standard menus header to display in the images menus rather than have the image label?

Thank you for your assistance.

paultech, 2020-07-10:
What portion are you having issues with? 

We document the use of JPEG byte arrays as images [here](https://github.com/ropg/M5ez#image-menus).

We do plan to add some simple convenience functions for loading of images from SD/SPIFFS.

pipy71, 2020-07-13:
Thanks for your assistance.
I was using the demo images menu and wanted to add a header to replace the white text label for the image.
But I figured out how to create a custom menu in the end.

closed 2020-07-13 by pipy71