M5ez/M5ez #76: ez.redraw() example needed

UT2UH, 2020-07-06:
Hello Rop!
Disparately need help with ez.redraw()  or some examples how to change submenu item’s caption from the redrawLoop() registered with ez.addEvent(redrawLoop). In my (too big) sketch one of the submenu item’s  advanced function assembles outgoing message and set pending message flag while radioLoop() waits for pending message flag, sends message if available,  checks if ACK message for this message received and if  acknowledgement message parsed successfully signals redrawLoop()  to redraw submenu’s items by ez.redraw() method.
Submenu and every event loops work as intended, messages are sent and received, signals passed between loops but there is no submenu items’ caption updates.
Thank you for awesome software and please teach us how to use ez.redraw with some example!

ropg, 2020-07-07:
I'm sorry I haven't had more time lately, have been working on some other projects. Hope to come back around here soon (but it will be a few weeks I am afraid) to read some issues and do some work...

paultech, 2020-07-08:
I use redraw without issue. I have however changed the rendering to use a back buffer to prevent flickering of text as its drawn.

`uint16_t updateBattery()
{
  ez.canvas.clear();
  ez.header.show("Debug");
  ez.canvas.font(mono6x8);
  ez.canvas.lmargin(10);

  ez.canvas.printf("APS:%.3fv\r\n", M5.Axp.GetAPSVoltage());
  ez.canvas.printf("USB:%.3fv/%.3fma\r\n", M5.Axp.GetVBusVoltage(), M5.Axp.GetVBusCurrent());
  ez.canvas.printf("BAT:%.3fv/%.3fma\r\n", M5.Axp.GetBatVoltage(), M5.Axp.GetBatCurrent());
  ez.canvas.printf("Bat:%:%d%%\r\n", ez.battery.getBatteryPerc());

  ez.redraw();
  return 500; // Change if you need to update a different rate
}

void sub_battery()
{
  ez.addEvent(updateBattery); // Add a event, this will cause m5ez to call the function as part of its internal loop.
  ez.buttons.wait("Exit"); //  Yield until exit button is pressed.
  ez.removeEvent(updateBattery);  // No longer update as we're exiting the menu.
}`

`  mainmenu.addItem("Debug", sub_battery);`

If you have any questions please let me know.

UT2UH, 2020-07-08:
Thank you, paultech!
I had a hope that with ez.redraw in the loop I can change for example the caption of "Debug" mainmenu item depending on data from received message parsed in this loop... But I think I can get what I need with printing to canvas in this item line... brute but if it will work it's OK. ) Thank you for the help!

paultech, 2020-07-10:
> Thank you, paultech!
I had a hope that with ez.redraw in the loop I can change for example the caption of "Debug" mainmenu item depending on data from received message parsed in this loop... But I think I can get what I need with printing to canvas in this item line... brute but if it will work it's OK. ) Thank you for the help!

Misunderstood,  try the drawFunction of the menu item? This allows you to take control of the drawing. 

https://github.com/ropg/M5ez#menus-all-the-functions-documented provides a myDrawFunction example.

closed 2020-07-14 by paultech