M5ez/M5ez #79: Event triggers every minute no matter what return value.

pipy71, 2020-07-09:
Hello, I am seeing some odd behaviour in that when returning from an event the next internal trigger is a minute later.

`ez.addEvent(myEvent);`   // trigger event now

`uint16_t myEvent(){
// do something
return 300000;    // set to trigger again in 5 minutes.
}
`
But it doesn't trigger in 5 minutes it triggers in 1 minute.
In fact whatever value I set 5 minutes or 1 hour it still triggers in 1minute.

Any ideas what I am doing wrong?

Thanks!

ropg, 2020-07-09:
16-bit unsigned integers go all the way up to 65 seconds and 535 milliseconds before wrapping back around to zero. M5ez's events are meant for things that need to happen frequently and need to live in the main loop, close to the metal where it's nice and warm.

If your events happen much less frequently, check out the [other events that are built into ezTime](https://github.com/ropg/ezTime#events).

closed 2020-07-09 by ropg

ropg, 2020-07-09:
I added a comment to this effect to the manual...

pipy71, 2020-07-10:
Thank you Rob, I really I appreciate your support.

With regards to the ezTime events can I set an event to just the hours of the day or do I have to specify the full date and time?
For example AddEvent("12:00");
I basically want the event to trigger the same time every day and not worry about setting the date.
Thanks!

ropg, 2020-07-10:
It says [here](https://github.com/ropg/ezTime#setevent) that you can supply either `TIME` (which is my shorthand for `time_t t = TIME_NOW, const ezLocalOrUTC_t local_or_utc = LOCAL`) or spell it out like `uint8_t hr, uint8_t min, uint8_t sec, uint8_t day, uint8_t mnth, uint16_t yr`.

So it follows that if you want to set it for the same date, you can supply the current values for date by inserting `day()`, `month()` and `year()` there.