M5ez/M5ez #83: Back Button

nak59, 2020-07-14:
Hello,
with code : 

void submenu_site(String SiteId , String NameSite) 
{
  ezMenu submenu_site(NameSite);
  submenu_site.buttons("up#Back#select##down#");
  submenu_site.txtSmall();
  submenu_site.addItem("Site Sumary");
  submenu_site.addItem("Access Points");
  //submenu_site.addItem("Switches");
  //submenu_site.addItem("Location");
  //submenu_site.addItem("Metrics");
  //submenu_site.runOnce();
  mainmenu.run();
  if (submenu_site.pickName() == "Access Points") {
    ez.msgBox("", "You pressed Access Points");
  }
  
}

pressing the back button does not move the previous menu but the main menu

vkichline, 2020-07-14:
Could you include the function implementing the previous menu?

nak59, 2020-07-14:
DeserializationError err = deserializeJson(doc, response);
              for (int i=0; i<sizeof(doc); i++)
                  {
                    String namesite =(doc[i]["name"].as<char*>());
                    String siteid =(doc[i]["id"].as<char*>());
                    String idnamesite = siteid +" | "+namesite;
                    if(namesite != "")
                    {
                        menusite.addItem(idnamesite);            
                    }
                  }
              }
            }
          numpage++;
           
          }
          //menusite.addItem("Back | Back to main menu");
          if (menusite.runOnce()) 
               {
                 String  id= menusite.pickName();
                 String namesite = menusite.pickCaption();
                 submenu_site(id , namesite);
               }

vkichline, 2020-07-14:
if (menusite.runOnce()) has completed, so menusite is terminated. Try:
while(menusite.runOnce()) {
    String  id= menusite.pickName();
   ...
}

nak59, 2020-07-14:
It's work :) Thks so much

paultech, 2020-07-15:
Glad to hear this is working :)

@vkichline as you mentioned in chat, This would be a good case to write an example for.

closed 2020-07-15 by paultech