M5ez/M5ez #84: Using the M5Core GPIO pins

misujr, 2020-07-15:
This is an amazing library ! Though trying to use it I can not figure out how to use the individual GPIO's on the side of the M5stack Core module with the basic back plate. 
If I want to do this with the M5Stack library I have to avoid using m5.begin() and I have to enable the functions I want ( like M5.Lcd.begin() ) in the setup loop. 
Basically the rest of the gpis's I use to control a stepper motor and read come interrupts. 
In case there is any suggestion from you I would be glad to hear it , otherwise this is not really an issue but more a question :-D . 

BR,
Misujr

ropg, 2020-07-15:
M5ez will execute `m5.begin()` for you when you execute `ez.begin()`, and you can start using the M5.Lcd commands directly after that. You might want to read the [manual section on the canvas](https://github.com/ropg/M5ez#canvas), "canvas" being the term we use for the area of the screen between the header (if displayed) and the buttons (if displayed). By using the `ez.canvas.top()` and `ez.canvas.bottom()` values, you can make sure you write to the screen without disturbing the rest of your UI.

Have fun!!

misujr, 2020-07-15:
Thank you for a such quick and complete reply :-) . I was trying to ask if I can use the library for creating a menu but without the m5.begin() to be called. The m5.begin() is the problem in my case so I need to work without calling that function :-) .

ropg, 2020-07-15:
Oh, now I see... So you are using pins that the display driver normally uses, or what? Which exact pins are we talking about and what part of the m5 library's init messes with what you want to do? And are you sure that whatever they do that you don't like cannot simply be undone?

Just looked at their code. This is [m5stack's `begin()`](https://github.com/m5stack/M5Stack/blob/master/src/M5Stack.cpp#L9):

```C++
void M5Stack::begin(bool LCDEnable, bool SDEnable, bool SerialEnable, bool I2CEnable) {
  // Correct init once
  if (isInited == true) {
    return;
  } else {
    isInited = true;
  }

  // UART
  if (SerialEnable == true) {
    Serial.begin(115200);
    Serial.flush();
    delay(50);
    Serial.print("M5Stack initializing...");
  }

  // LCD INIT
  if (LCDEnable == true) {
    Lcd.begin();
  }

  // TF Card
  if (SDEnable == true) {
    SD.begin(TFCARD_CS_PIN, SPI, 40000000);
  }

  // TONE
  // Speaker.begin();

  // Set wakeup button
  Power.setWakeupButton(BUTTON_A_PIN);

  // I2C init
  if (I2CEnable == true) {
    Wire.begin(21, 22);
  }

  if (SerialEnable == true) {
    Serial.println("OK");
  }

  // if use M5GO button, need set gpio15 OD or PP mode to avoid affecting the wifi signal  
  pinMode(15, OUTPUT_OPEN_DRAIN);
}
```

With the [corresponding header file](https://github.com/m5stack/M5Stack/blob/master/src/M5Stack.h#L125) defining:

```C++
void begin(bool LCDEnable = true, bool SDEnable = true, bool SerialEnable = true, bool I2CEnable = false);
```

We should be defining our `ez.begin()` function the same way and just pass these values along, which we presently do not do. I'll make note to fix this, so you can decide which parts you want to enable.

misujr, 2020-07-15:
very likely the serial is my problem but not sure if any others will clash. I am using pins 3,1,16,17,2 . I am sorry that I am unable to bring much value with my response but I am learning this part now :-) . I am using the pins in the red rectangle basically. 
[image]
LE: Using you suggestion I modified the m5stack library locally and now it all works. Serial was my problem here. I am sorry to have bothered you with this thing and I am super grateful for your help in this matter. :-D

ropg, 2020-07-15:
Init without serial is a perfectly normal thing to want. And the m5stack library supports this with its optional arguments to `begin()`. M5ez will support these same arguments.

I just fixed it in [this branch](https://github.com/ropg/M5ez/tree/fix_begin) (or [zip file](https://github.com/ropg/M5ez/archive/fix_begin.zip)). You could do us a favor by testing it. Change back the m5stack library, use this new version of M5ez and issue

```C++
ez.begin(true, true, false, false);
```

That should disable the serial port. It'll be part of the next version of M5ez, [manual updated too](https://github.com/ropg/M5ez/tree/fix_begin#how-it-all-works).

misujr, 2020-07-15:
I have just downloaded the fix in the zip format and I have : 
- Reverted my M5stack library to original
- tested with the serial argument ( 3rd) false => all works perfect 
- tested with the serial argument true => not working - expected
- Tested again with serial argument false => all working perfect
Thank you very much for such a quick fix and I hope this was constructive :-D . 
I will use the library in this form until the update gets pushed thru Arduino but it is just amazing :-D .

closed 2020-07-15 by misujr

ropg, 2020-07-15:
OK, thanks. Do note that at some undetermined point in the future we may decide to change things and make it such that users have to first call `m5.begin()` again, as we move a bit from programming environment to interface library. But if that happens, it'll be part of a major change, possibly when we go to 3.0.0.

misujr, 2020-07-15:
Absolutely no problem. I will keep an eye on the development of the library
and update my code to make sure it will work :-D .