M5ez/M5ez #88: Access to calling menu functions from ez.addEvent function

UT2UH, 2020-07-17:
Please add the access to calling menu functions from a function registered by ez.addEvent like it’s available in item’s advancedFunction  and drawFunction.
An opportunity to update menu item’s caption will allow to combine real-time display of some parameters and their control – for example, an event function redraws menu item’s caption to “FeatureName/t  ON” to display received current state of some remote device while advanced function of this item allows to toggle the device state to OFF and send the corresponding command to it.

paultech, 2020-07-17:
Understand the reasoning behind that, we are having a rethink of the API for the next big release in order to make these tasks easier and follow a more event-based callback architecture.

@ropg @vkichline We'll need to make some minor improvements to the current event system until we can break the API. 
Seems fair to add a addEvent() that takes a funcptr passing in the calling menu? No objections here other than regular dislike adding to the API we are about to change.

UT2UH, 2020-07-17:
@paultech, thank you! It will be really a great improvement!

ropg, 2020-07-17:
I think we should feel very free to add non-breaking changes, since I sense think it may be a while before we have a fine-grained enough idea of what M5ez will turn into and then will have written everything, and then feeling it's tested by enough people to be the replacement. 

As for passing the menu to events, we may want to then have a menu function to exit the menu programmatically, with some return value for `.runOnce()`. Maybe `.pick` with an argument supplied? This could be an option that is not otherwise in the menu, and would likely need a variable.

Come to think of it: `.exit(<returnval>)` may be better because one would want to preserve the 'unpicked' cursor position in `.pick()` so you could just back to the menu and possible select the right option... Or?

UT2UH, 2020-08-10:
Hello Devs!
Any chance to see such improvements in the next release? Asking just to plan further development steps.

vkichline, 2020-08-10:
Hello, sorry this has been idle. I am looking into it now, but am not certain. Currently events have no way to know what the current menu is, and the menu is free to change while the event exists. I'm going to experiment with revealing the menu and see if it's reliable. If this can be done 100% safely I will try to get it in.

vkichline, 2020-08-10:
I've experimented with this and have added a pointer to the current menu in the M5ez class (in test code, not checked in yet.) It may be NULL at times, so you'll have to check it before it's used (as you would have to with the callback.) This approach leaves us with a lot more flexibility going forward. You would access the current menu by just doing:
```
uint16_t evt() {
  Serial.printf("Current menu pointer is: %d\n", ez.currentMenuPtr);
  return 100;
}
```
Would this serve your needs?

UT2UH, 2020-08-10:
> 
> 
> and the menu is free to change while the event exists. 
Hello!
May be some details will be useful.
Right now I lowered the canvas one more row down under the header and draw all received ACKs from events to this status line. Below are various menus as in original code. But even now I do that with several events having almost the same code but each one only for the menu I am.  And the functions I need from the calling menu to replace this output to my status line are callingMenu->setCaption and callingMenu->deleteItem.

UT2UH, 2020-08-10:
Sorry - * each one only for the menu I am in.

UT2UH, 2020-08-10:
I am ready to test your experimental code on my rather specific hardware and to prepare some example for native M5Stack hardware on the weekend (need to pick up M5Stack from a friend).

vkichline, 2020-08-10:
The code is in a fork right now, not even a PR yet; the quickest way to test it would be to download it from https://github.com/vkichline/M5ez/tree/menu_ref.  I changed `ez.currentMenuPtr` to `ez.currentMenu` for consistency.
Thanks for providing details, that's just what I need, but I'm not certain I understand correctly.  Is there any chance I could look at your source code? It may help me identify another approach.
There are a couple other way we could try to address this from the M5ez side: passing a general-purpose value to addEvent (which could be a menu handle, or other storage) or creating a class of events which only run when a specific menu is active.

UT2UH, 2020-08-11:
@vkichline 
Thank you! Well done, both `ez.currentMenu->setCaption()` and `ez.currentMenu->deleteItem()` work flawlessly! 
Menu are "live" now - finished processes removed from main menu on received acknowlegements and received variables update processes item's captions to show their current state without touching a button!
I had to add a screen wakeup function not to miss incoming updates (M5.Lcd.writecommand()s below also save some battery capacity):
`void ezBacklight::wakeup() {
	if(_backlight_off){
		M5.setLcdBrightness(_lcdbrightness, _lcdphase0);
		M5.Lcd.writecommand(0x29); // ILI9341_DISPON
		M5.Lcd.writecommand(0x11); // ILI9341_SLPOUT
		_backlight_off = false;
	}
	activity();
}`
and to work with my modified backlight loop:
`uint16_t ezBacklight::loop() {
	if (!_backlight_off && _inactivity) {
		if (millis() > _last_activity + 1000 * _inactivity) {
			if(!_backlight_off){
				M5.setLcdBrightness(0xF, HIGH);
				M5.Lcd.writecommand(0x28); // ILI9341_DISPOFF
				M5.Lcd.writecommand(0x10); // ILI9341_SLPIN
				_backlight_off = true;
			}
			while (true) {
				if (m5.BtnA.wasPressed() || m5.BtnB.wasPressed() || m5.BtnC.wasPressed() || m5.BtnD.wasPressed() || m5.BtnE.wasPressed() || m5.BtnF.wasPressed()) break;
				ez.yield();
				delay(10);
			}
			ez.buttons.releaseWait();	// Make sure the key pressed to wake display gets ignored
			if(_backlight_off){
				M5.setLcdBrightness(_lcdbrightness, _lcdphase0);
				M5.Lcd.writecommand(0x29); // ILI9341_DISPON
				M5.Lcd.writecommand(0x11); // ILI9341_SLPOUT
				_backlight_off = false;
			}
			activity();
		}
	}
	return 1000;
}`
A lot of things now have to be redone completely but I am more than happy to this extra job - now it's quite different and very dynamic environment!
My thanks to all developers of this excellent library!

vkichline, 2020-08-11:
I will expedite getting this added, pending code review. Expect it to take a few days, but the code you installed is identical otherwise and is fine to use for now.

UT2UH, 2020-08-11:
Great, it will be a benefit for user community! For me the time of upstream update is not so important as I can not use original M5ez code on my custom hardware - I already merged your update with my fork. Thank you!

vkichline, 2020-08-12:
Fixed. UT2UH: note that the interface changed slightly after your testing.

closed 2020-08-12 by vkichline

UT2UH, 2020-08-13:
So, with
```
if(cur_menu->getTitle() == "Desired Menu Title"){
} else {
}
```
now, I can do menu branching directly in events? Nice! ) Thank you!