M5ez/M5ez #92: Can I adapt this to other screens, in particular M5StackC?

dpharris, 2020-07-29:
Hi -
Fantastic library.  Is it possible to adapt this to the M5StackC and other ESP32 boards with LCDs.  Is it as simple as defining the screen size and possibly rotating it?  
Thabks
David

paultech, 2020-07-29:
Hello @dpharris 

We actually have upcoming platform abstraction support that includes the M5StackC! Other ESP32 may be possible in the future as the abstraction is designed to be flexible.

From experience on having this run on M5StickC you have to change the follow

* Include M5StickC.h vs M5Stack.h
* Change LCD_W LCD_H
* The power functions between M5Stack and M5StickC are not equal, you'll need to either disable EZ_BATTERY or port the functions (Different power management IC's)
* LCD Brightness, you control it via the power management IC (setting the backlight voltage) vs directly on the LCD class. You'll need to port to disable brightness control.

Let me know if you need any assistance in testing, we hope to publish our roadmap this week that outline these new features.

dpharris, 2020-07-29:
Excellent news.  Thank you.
I am expecting my M5StackCs this week, so the timing is perfect for me.

vkichline, 2020-08-16:
Hello dpharris; we are preparing for a minor release of M5ez, so I am closing this issue and moving your request to the product backlog. I'm afraid the M5StickC support will be in a major release, whose date isn't scheduled yet. Follow this project for updates.

closed 2020-08-16 by vkichline

dpharris, 2020-08-16:
C'est la vie.  Thanks for the note.