M5ez/M5ez #93: Set time and date manually from clock settings menu

pipy71, 2020-08-03:
Hi, I am a total newbie so I do apologise if this is a stupid question.

But Is there a menu option for the user to set the time and date manually from a menu screen in the clock settings?

I want the option for when the device is offline and not being connected to the internet, but I still need the time and date to be logged.

if there isn't a menu option my guess is I will have to make one, but I have no idea how to begin. so any help would be really appreciated. 

Thank you.

ropg, 2020-08-03:
There is no such option, but one could be made.

I fear the bigger problem is going to be "clock drift": the clock won't be accurate after a while. Also: if you are not going to keep it powered on at all times, you have to resort to the ESP32's deep sleep functionality, which can keep time but I have no idea how.

pipy71, 2020-08-03:
Thank you for the quick reply and clear explanation.

Okay in which case I think I would need the user to switch on the M5Stack and then be forced to connect the wifi before being allowed to use the M5. Naturally on future switch on the autconnect will remember the wifi.

What would be the best way do you think to make sure the users connect the wifi for the first time?
Can I, for example, check autoconnect state and then switch directly to the wifi settings screen if no autoconnect set, if so how do i do this?

Or do you have another better idea on locking functionality until after wifi is configured?

Thanks for your wonderful support!

UT2UH, 2020-08-03:
@pipy71,
Here is my code for manual time entry  "as is". Sorry, I use M5ez forked for pre-M5Stack 6-button hardware so you will have to fix msgBox buttons back... 
Clock drift is nothing if there is no WiFi at all, there should be some way to set the system time.

`bool mngSystemTime(ezMenu* callingMenu) {    
  tmElements_t screenTime;
  time_t tEntered;
  
  String disp_val;
  uint16_t highlight_color = TFT_RED;
  ez.header.show("TIME");   
  if (timeStatus() == timeSet) {   
    breakTime(UTC.now(), screenTime); 
    screenTime.Year -= 30;
    disp_val = "Enter system time | in dd.mm.yy hh:mm format";
  } else {
    screenTime.Hour   = 12;
    screenTime.Minute = 0;
    screenTime.Day    = 1;
    screenTime.Month  = 4;
    screenTime.Year   = 20;
    disp_val = "The clock stopped | Enter system time | in dd.mm.yy hh.mm format";
  }
  uint8_t elementSet = TM_DAY;
     
  while (true) {     
    ez.msgBox("TIME", disp_val, " Back| # +| # select| # GPS| # -| # OK| # # # # # ++| # # # --| # ", false, &FreeMono9pt7b);
    ez.canvas.font(&FreeMonoBold18pt7b);
    M5.lcd.fillRect (0, ez.canvas.bottom() - 50, TFT_W, 40, ez.theme->background); 
    ez.canvas.pos(13, ez.canvas.bottom() - 50);
    
    ez.canvas.color(elementSet == TM_DAY    ? highlight_color : ez.theme->msg_color);            
    ez.canvas.print(zeropad((uint32_t)screenTime.Day, 2));
    ez.canvas.print(".");
    ez.canvas.color(elementSet == TM_MONTH  ? highlight_color : ez.theme->msg_color);            
    ez.canvas.print(zeropad((uint32_t)screenTime.Month, 2));
    ez.canvas.color(ez.theme->msg_color);        
    ez.canvas.print(".");      
    ez.canvas.color(elementSet == TM_YEAR   ? highlight_color : ez.theme->msg_color);
    ez.canvas.print(zeropad((uint32_t)screenTime.Year, 2));
    ez.canvas.color(ez.theme->msg_color);        
    ez.canvas.print(" ");
    ez.canvas.color(elementSet == TM_HOUR   ? highlight_color : ez.theme->msg_color);  
    ez.canvas.print(zeropad((uint32_t)screenTime.Hour, 2));
    ez.canvas.color(ez.theme->msg_color);        
    ez.canvas.print(":");
    ez.canvas.color(elementSet == TM_MINUTE ? highlight_color : ez.theme->msg_color);          
    ez.canvas.print(zeropad((uint32_t)screenTime.Minute, 2));
    String b = ez.buttons.wait();
    if (b == "select") elementSet == 4 ? elementSet -= 4 : elementSet++ ; 
    switch (elementSet) {
      case TM_DAY:
        if (screenTime.Day > days[screenTime.Month-1]) screenTime.Day = days[screenTime.Month-1];
        if (b == "-" && screenTime.Day >  1) screenTime.Day--;
        if (b == "+" && screenTime.Day < days[screenTime.Month-1]) screenTime.Day++;
        if (b == "--") {
          if (screenTime.Day < 10) {
            screenTime.Day = 0;
          } else {
            screenTime.Day -= 10;
          }
        }
        if (b == "++") {
          if (screenTime.Day > 20) {
            screenTime.Day = days[screenTime.Month-1];
          } else {
            screenTime.Day += 10;
          }
        }
        break;
      case TM_MONTH: 
        if (((b == "-") || (b == "--")) && (screenTime.Month >  1)) screenTime.Month--;
        if (((b == "+") || (b == "++")) && (screenTime.Month < 12)) screenTime.Month++;
        if (screenTime.Day > days[screenTime.Month-1]) screenTime.Day = days[screenTime.Month-1];
        break;
      case TM_YEAR: 
        if (((b == "-") || (b == "--")) && (screenTime.Year > 20)) screenTime.Year--;
        if (((b == "+") || (b == "++")) && (screenTime.Year < 36)) screenTime.Year++;
        break; 
      case TM_HOUR: 
        if (b == "-" && screenTime.Hour >  0) screenTime.Hour--;
        if (b == "+" && screenTime.Hour < 23) screenTime.Hour++;
        if (b == "--") {
          if (screenTime.Hour < 3) {
            screenTime.Hour = 0;
          } else {
            screenTime.Hour -= 3;
          }
        }
        if (b == "++") {
          if (screenTime.Hour > 21) {
            screenTime.Hour = 0;
          } else {
            screenTime.Hour += 3;
          }
        }
        break;
      case TM_MINUTE: 
        if (b == "-" && screenTime.Minute >  0) screenTime.Minute--;
        if (b == "+" && screenTime.Minute < 59) screenTime.Minute++;
        if (b == "--") {
          if (screenTime.Minute < 10) {
            screenTime.Minute = 0;
          } else {
            screenTime.Minute -= 10;
          }
        }
        if (b == "++") {
          if (screenTime.Minute > 50) {
            screenTime.Minute = 59;
          } else {
            screenTime.Minute += 10;
          }
        }
        break;          
    }    
    if (b == "OK") {
      screenTime.Year  += 30;
      screenTime.Second = 00;      
      tEntered = makeTime(screenTime);
      setRtcEpoch(tEntered);        // use the time_t value to ensure correct weekday is set
      UTC.setTime(getRtcEpoch());     
      break;
    } else if (b == "Back") {
      break;
    }
  }
  return true;
}
`

pipy71, 2020-08-03:
Thanks so much for the code, but when I said I was a newbie I really meant it! 
I am so sorry to bother you again but I have no idea where to put this code?
I tried putting this in my sketch but it complained about the very first line when I tried to call it from my menu item and also about the lines such as TM_DAY not being declared in this scope.

UT2UH, 2020-08-03:
Add to your menu time entry item
`mainmenu.addItem("Enter system time", NULL, mngSystemTime);`

and add 
`	typedef enum {
		TM_DAY    = 0,
		TM_MONTH  = 1,
		TM_YEAR   = 2,
		TM_HOUR   = 3,
		TM_MINUTE = 4
	} tmEntered_t;`

pipy71, 2020-08-03:
Thanks, one more error...

'days' was not declared in this scope.

UT2UH, 2020-08-03:
`uint8_t days[12] = {31,29,31,30,31,30,31,31,30,31,30,31};`

pipy71, 2020-08-03:
Sorry also setRtcEpoch was not declared...

 `setRtcEpoch(tEntered);`        // use the time_t value to ensure correct weekday is set
     
and \i am guessing the next line will also throw an error...

 `UTC.setTime(getRtcEpoch());`

UT2UH, 2020-08-03:
try to replace
`      setRtcEpoch(tEntered);        // use the time_t value to ensure correct weekday is set
      UTC.setTime(getRtcEpoch()); ` 
just with
`      UTC.setTime(tEntered); `

vkichline, 2020-08-16:
Hello, pipy71. We are preparing for a new minor release and are closing outstanding issues in preparation.
Thanks for helping out UT2UH!
It sounds like this issue has been resolved, but I'm going to place it on the project backlog to revisit in future major releases.
Thanks! Feel free to reopen the issue if you feel that it needs further attention.

closed 2020-08-16 by vkichline