M5ez/M5ez #94: 2.1.2 release from Arduino IDE library manager in consistent

cwerfel, 2020-08-07:
First, thank you for this amazing contribution! Awesome bit of code for the M5. Second, this could be user error but when I installed 2.1.2 from the Arduino IDE library manager, the codeline I got did not include the clock code. I went ahead and pulled the src for 2.1.2 and installed manually. I'm not very experienced with the plumbing behind the Arduino IDE and publishing libraries so this may be a non-issue. Thanks!

vkichline, 2020-08-07:
We are preparing to update the release and plan to keep it up to date going forward. Glad you worked through the issue!

vkichline, 2020-08-17:
2.3.0 has been released. The Arduino Library Manager has not updated itself yet, but should soon.

closed 2020-08-17 by vkichline