M5ez/M5ez #107: Add selected to runOnce

benhul, 2020-09-21:
Your library works great. I use it to set the options for my program. 

Maybe you can add an optional selected field to the .runOnce function.

[image]

I use a menu to set preferences and I use a \t to show the current value to the right on  the same line. After clicking +, 5 is added to the current value. I would like to renew the menu with the new value (which I can) but the same (second) row selected (which I can't).

I made the change for myself (it's only 4 lines of code::smile: ) but others could benefit. 

``` cpp
int16_t ezMenu::runOnce(int selected) {
	_selected = selected;           // or selected - 1 to make selected 1 based
	return runOnce();
}
```

vkichline, 2020-09-24:
Good suggestion, @benhul; it would be helpful to be able to set the menu selection.
Recently we added a way to access the current menu from an event; this won't help with your issue, but it makes me think it would be better to add an `ezMenu::setSelection(int n)` instead; then the selection could be altered by a periodic event as well as in a call to `runOnce()`. Maybe it would be `setPick(int n)` as that seems more consistent if the function uses 1-based indexes for the items as everywhere else.
I think all you would have to do (if this function were added) would be:

```
    menu.setPick(2);
    menu.runOnce();
```

Would it work for you if it were implemented this way?

benhul, 2020-09-25:
Good name `setPick` and indeed it's better to have it one based. I (would) use:

```c++
int current_row = 0;
//...
    menu.setPick(current_row);
    runOnce();
    current_row = menu.pick();
```
Have fun.

UT2UH, 2020-09-25:
You can write color managers like this 
```
volatile uint8_t colorValue;
bool mngColor(ezMenu* callingMenu){
  colorValue += 5;
  callingMenu->setCaption("color", "Color\t" + String(colorValue));
  return true;
}
```

benhul, 2020-09-26:
You're right, missed that one...

MartinNohr, 2021-04-07:
I could use the setPick() idea as well. Any idea when this will be impemented?
Thanks.

LennartHennigs, 2021-04-07:
Hey, I added these in my fork because I am building something similar:

https://github.com/M5ez/M5ez/compare/master...LennartHennigs:master

MartinNohr, 2021-04-07:
I don't think your code is completely correct.
void ezMenu::setItem(int index) {
	_selected = index;
}
What happens if the menu was larger and had been scrolled? It doesn't display correctly in that case.