M5ez/M5ez #108: How do I show the new Batterylevel widget

rhannink, 2020-10-12:
Hello,

I am trying to display the new batterylevel widget, together with the Wifi and time widget in the righthand corner of the header. The time and wifi strength show by default. How do I add the batterylevel widget?

It looks like the documentation is not mentioning the batterylevel yet.

Gr. Remco

rhannink, 2020-10-12:
Hello,

In the meantime I managed via two ways, but wondering what the "Royal" way is:

- Run the m5ez demo once and set batterylevel to "on" via menu. This stores the preference on FS
- Change the preferences on SPDIFFS (prefs.putBool("battery_icon_on", true); 

Gr. Remco

vkichline, 2020-10-12:
Hi, @rhannink!
I would call the first way the "Royal" way, the way it was intended to be activated. The feature is controlled by a setting preference, and either of the two ways you accomplished this sets the setting to "on". The "ez" way would be to include the built-in settings menu. But there's absolutely no need; it all depends on your needs.
A third way would be to be to manipulate the settings with a separate program. When I was testing clearing settings, or when I set them to weird values for testing, I use an app I wrote to save/restore settings to a file. You can edit the json file with any editor. You can find the app [here](https://github.com/vkichline/M5ezSettingsStorage).

LennartHennigs, 2022-04-10:
this is what I used in my `setup()`

```
  Preferences prefs;
  prefs.begin("M5ez");
  prefs.putBool("battery_icon_on", true);
  prefs.end();
```

not pretty, but it works.