M5ez/M5ez #109: What's Going On with M5ez?

vkichline, 2020-10-13:
Right now Rop is working hard on adding some much needed Touch functionality to the M5Core2 library.
I'm doing reviews and developing test apps.
This is critical right now, because the 0.0.1 release of M5Core2 was far from compatible with M5ez, and the Touch UI was incomplete.
Once this work is done ("Real Soon Now") work will resume on M5ez and will focus on making the library available on the Core2, the M5Stick and M5Stick PLUS as well.
Work hasn't paused, it's just focused elsewhere for now.

If you've been playing with the M5Core2, do you see yourself using some form of M5ez there in the future?
Are there any new features you like to see to make the experience better on the Core2?

pipy71, 2020-12-03:
Hello, 

Thank you for the update, I am really looking forward to seeing M5ez on the M5core2.

Do you have an expected release date yet?
Are you going to make use of the RTC in the M5Core2?
Is there a beta version available for testing?

Thanks

ropg, 2020-12-03:
There is something that is more of a very early alpha. And it's turned into more of a lightweight Qt than the original M5ez. Which is a good thing when you have a touch screen.

If you can deal in not having documentation and just a bunch of examples, check out https://github.com/M5ez/Core2ez . Expect things to change...

LyleDodge, 2021-06-16:
@ropg when you get to the point of wanting testing feedback, I'm happy to help. I've been working extensively with M5ez on an M5Stack Fire over just the last few weeks but getting pretty deep in and loving it. My Core2 is on the way and I'll give the Core2ez library a spin once it comes in.

Thanks for all you're doing for the community!

dizcza, 2023-04-07:
Any plans to support arduino-esp v2.x?

With the latest PlatformIO (arduino v2.0.7), I'm getting

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7) 
 - tool-esptoolpy @ 1.40500.0 (4.5.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5ez @ 2.3.0+sha.fa34e2d
|   |-- M5Stack @ 0.4.3
|   |   |-- Wire @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SD @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |   |-- SPIFFS @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |-- ezTime @ 0.8.3
|   |   |-- EEPROM @ 2.0.0
|   |   |-- Preferences @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- Preferences @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- ezTime @ 0.8.3
|   |-- EEPROM @ 2.0.0
|   |-- Preferences @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- EEPROM @ 2.0.0
|-- Preferences @ 2.0.0
|-- SD @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio/build/frogboard/lib569/ezTime/ezTime.cpp.o
Compiling .pio/build/frogboard/lib7b6/M5ez/M5ez.cpp.o
Compiling .pio/build/frogboard/src/main.cpp.o
Building .pio/build/frogboard/bootloader.bin
esptool.py v4.5
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generating partitions .pio/build/frogboard/partitions.bin
Compiling .pio/build/frogboard/lib3a0/Wire/Wire.cpp.o
Compiling .pio/build/frogboard/lib86e/FS/FS.cpp.o
Compiling .pio/build/frogboard/lib86e/FS/vfs_api.cpp.o
.pio/libdeps/frogboard/M5ez/src/M5ez.cpp: In lambda function:
.pio/libdeps/frogboard/M5ez/src/M5ez.cpp:1133:38: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
    if(WIFI_REASON_ASSOC_FAIL == info.disconnected.reason) {
                                      ^~~~~~~~~~~~
[... 26 lines cut ...]
/home/dizcza/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:159:21: note:   conversion of argument 1 would be ill-formed:
.pio/libdeps/frogboard/M5ez/src/M5ez.cpp:1420:29: error: invalid conversion from 'void (*)(arduino_event_id_t, system_event_info_t)' to 'WiFiEventCb' {aka 'void (*)(arduino_event_id_t)'} [-fpermissive]
      WiFi.onEvent(_WPShelper);
                             ^
In file included from /home/dizcza/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:28,
                 from /home/dizcza/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:32,
                 from .pio/libdeps/frogboard/M5ez/src/M5ez.h:36,
                 from .pio/libdeps/frogboard/M5ez/src/M5ez.cpp:1:
/home/dizcza/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:161:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventSysCb, arduino_event_id_t)' <near match>
     wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
/home/dizcza/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:161:21: note:   conversion of argument 1 would be ill-formed:
.pio/libdeps/frogboard/M5ez/src/M5ez.cpp:1420:29: error: invalid conversion from 'void (*)(arduino_event_id_t, system_event_info_t)' to 'WiFiEventSysCb' {aka 'void (*)(arduino_event_t*)'} [-fpermissive]
      WiFi.onEvent(_WPShelper);
                             ^
.pio/libdeps/frogboard/M5ez/src/M5ez.cpp: In static member function 'static void ezWifi::_WPShelper(arduino_event_id_t, system_event_info_t)':
.pio/libdeps/frogboard/M5ez/src/M5ez.cpp:1467:17: warning: comparison between 'enum arduino_event_id_t' and 'enum system_event_id_t' [-Wenum-compare]
    if (event == SYSTEM_EVENT_STA_WPS_ER_PIN) {
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Archiving .pio/build/frogboard/lib3a0/libWire.a
Indexing .pio/build/frogboard/lib3a0/libWire.a
Compiling .pio/build/frogboard/lib79f/SPI/SPI.cpp.o
*** [.pio/build/frogboard/lib7b6/M5ez/M5ez.cpp.o] Error 1
======================================================================= [FAILED] Took 3.74 seconds =======================================================================
```

a4x4kiwi, 2023-04-11:
Yes, I would be nice if someone smarter than me could bring this project forward to support WiFi and various other patches that have been provided like the animated charging indicator.