M5ez/M5ez #110: How not to show buttons

youleaf, 2020-11-04:
Can't figure out how not to show buttons.

My app needs to display data in full screen (no header / no buttons). The user can choose the displayed data from a main menu. When data are displayed, pressing any button should bring the user back to the main menu.

I successfully removed the buttons by commenting out `ez.buttons.show()`, but I can no more exit from the `while (btnpressed == "")` loop I use to update/display current data.

Which is the correct way to implement such scenario?

Thank you for your awesome project!

ropg, 2020-11-04:
No time to test it right now. But this should solve one half of your problem: displaying a message without buttons: 

`ez.msgBox("Some header", "Some message, "", false);`

That should (untested) display a full screen message with no buttons and not wait for anything after. Now the next problem is that M5ez then doesn't know about the buttons, so you cannot use `ez.buttons.poll`. I think it makes most sense to create this functions in your own code:

```c++
void anyButton() { while (!(M5.BtnA.wasPressed() || M5.BtnB.wasPressed() || M5.BtnB.wasPressed()))  ez.yield(); }
```

(The function first waits for buttons to be released so it doesn't immediately exit if a key is still pressed when the user gets that message.)

youleaf, 2020-11-04:
Thank you @ropg!

Starting from your suggestion, I came up with this; I'm sharing it here in case it's useful for someone.

```cpp
void test_display()
{
  ez.canvas.reset();
  ez.canvas.scroll(false);
  ez.buttons.clear();

  // check actual screen size (for testing only)
  Serial.print("ez.canvas.height: ");
  Serial.println(ez.canvas.height()); // should be 240
  Serial.print("ez.canvas.bottom: ");
  Serial.println(ez.canvas.bottom()); // should be 239

  while (!(M5.BtnA.wasPressed() || M5.BtnB.wasPressed() || M5.BtnC.wasPressed()))
  {
    ez.buttons.poll();
    //
    // on screen data updating goes here...
    //
    delay(100);
  }
  // de-bounce button press
  delay(100);
  ez.buttons.poll();
  ez.canvas.reset();
}
```

closed 2020-11-04 by youleaf