M5ez/M5ez #131: cannot connect to raw.githubusercontent.com for OTA

skysingh, 2022-03-31:
We have using M5ez for years and also the OTA capability you so graciously provided. I used github as the repository and all worked fine for years. Now we get “Connection to raw.githubusercontent.com failed” and cannot do OTA !

Any idea what is going on and how we fix/troubleshoot ? We have lots of devices in the field and we need the OTA capability.

MartinNohr, 2022-06-05:
I just noticed the same problem. Has something on github changed perhaps?

skysingh, 2022-06-05:
No idea. If you figure it out tell me. We are emailing uploads to customers to resolve the mess until we transition to another OTA method.

 

*	Vijay

MartinNohr, 2022-06-06:
I'm thinking that GitHub has changed something in their URL's that is breaking this. I'll see if I can figure it out. If not I'm also pursuing using some other OTA code. Fortunately I haven't deployed my system yet, but in another month this would be a much larger problem. I'll let you know what I did.
Martin

MartinNohr, 2022-06-06:
OK, I fixed the example. You have to regenerate the .h file using get_cert. Probably the github certificate changed so the old .h didn't work. I also noticed that the documentation has a minor error. I says you need opensll, but that is of course openssl.
I haven't tried my own updater yet, but I suspect I need to update the .h file. This is of course a huge problem if you have deployed systems. You need to install newly compiled binary using the new .h file before you can OTA new firmware! I'm not enough of an expert on certificates to know if there is some way around this. One kludgy way might be to include the .h contents on an SD card and read it in instead of compiling it in. There is an SD reader so that could be done. I don't know how often this will be a problem with needing new certificate info. Any experts out there with better ideas?

ropg, 2022-06-07:
Certificate expiry and embedded devices: an industry-wide unsolved problem. See for example [this news story](https://hackaday.com/2022/05/30/expired-certificate-causes-german-payment-meltdown/) that is just a week old.

A solution is to put your updates on a server/VPS you control and use a certificate with your own root that expires in the year 7000. Here's a [thread](https://threadreaderapp.com/thread/1532058883908198401.html) about why certificate expiry is overrated anyway. But then use that cert only on an update binaries subdomain that doesn't need to to ever be served to actual browsers, because they won't like you using your own root cert.

closed 2022-06-07 by ropg