M5ez/M5ez #134: COMPILATION ERROR

udrojunior, 2023-10-26:
I'm having the following error when compiling, from what I understand it's a matter of incompatibility, but I've already updated everything and I'm still having this error: 
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp: In lambda function:
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1133:38: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
    if(WIFI_REASON_ASSOC_FAIL == info.disconnected.reason) {
                                      ^~~~~~~~~~~~
                                      eth_connected
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp: In static member function 'static void ezWifi::begin()':
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1139:19: error: 'SYSTEM_EVENT_STA_DISCONNECTED' is not a member of 'arduino_event_id_t'
   }, WiFiEvent_t::SYSTEM_EVENT_STA_DISCONNECTED);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp: In static member function 'static bool ezWifi::_connection(ezMenu*)':
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1410:13: error: 'struct esp_wps_config_t' has no member named 'crypto_funcs'
      config.crypto_funcs = &g_wifi_default_wps_crypto_funcs;
             ^~~~~~~~~~~~
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1410:29: error: 'g_wifi_default_wps_crypto_funcs' was not declared in this scope
      config.crypto_funcs = &g_wifi_default_wps_crypto_funcs;
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1410:29: note: suggested alternative: 'g_wifi_default_wpa_crypto_funcs'
      config.crypto_funcs = &g_wifi_default_wps_crypto_funcs;
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                             g_wifi_default_wpa_crypto_funcs
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1420:29: error: no matching function for call to 'onEvent(void (&)(arduino_event_id_t, system_event_info_t))'
      WiFi.onEvent(_WPShelper);
                             ^
In file included from C:\Users\Suporte T.I\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi\src/WiFiSTA.h:28,
                 from C:\Users\Suporte T.I\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi\src/WiFi.h:32,
                 from c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src/M5ez.h:36,
                 from c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1:
C:\Users\Suporte T.I\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi\src/WiFiGeneric.h:159:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventCb, arduino_event_id_t)' <near match>
     wifi_event_id_t onEvent(WiFiEventCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
C:\Users\Suporte T.I\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi\src/WiFiGeneric.h:159:21: note:   conversion of argument 1 would be ill-formed:
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1420:29: error: invalid conversion from 'void (*)(arduino_event_id_t, system_event_info_t)' to 'WiFiEventCb' {aka 'void (*)(arduino_event_id_t)'} [-fpermissive]
      WiFi.onEvent(_WPShelper);
                             ^
In file included from C:\Users\Suporte T.I\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi\src/WiFiSTA.h:28,
                 from C:\Users\Suporte T.I\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi\src/WiFi.h:32,
                 from c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src/M5ez.h:36,
                 from c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1:
C:\Users\Suporte T.I\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi\src/WiFiGeneric.h:161:21: note: candidate: 'wifi_event_id_t WiFiGenericClass::onEvent(WiFiEventSysCb, arduino_event_id_t)' <near match>
     wifi_event_id_t onEvent(WiFiEventSysCb cbEvent, arduino_event_id_t event = ARDUINO_EVENT_MAX);
                     ^~~~~~~
C:\Users\Suporte T.I\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi\src/WiFiGeneric.h:161:21: note:   conversion of argument 1 would be ill-formed:
c:\Users\Suporte T.I\Documents\Arduino\libraries\M5ez\src\M5ez.cpp:1420:29: error: invalid conversion from 'void (*)(arduino_event_id_t, system_event_info_t)' to 'WiFiEventSysCb' {aka 'void (*)(arduino_event_t*)'} [-fpermissive]
      WiFi.onEvent(_WPShelper);
                             ^

exit status 1

Compilation error: exit status 1

a4x4kiwi, 2023-12-21:
Disable wifi in the library, then it may work. The ez library is not compatible with recent esp32 wifi.

ewoodier, 2025-02-11:
> Disable wifi in the library, then it may work. The ez library is not compatible with recent esp32 wifi.

How do you disable WiFi in the library? I am still having the same issues

a4x4kiwi, 2025-02-12:
Edit the m5ez.h file to enable and disable features.

Line 11.

// Turn this off to compile without WiFi (no) OTA updates, no clock)
#define M5EZ_WIFI